Mattiwatti/EfiGuard #1: Re-enabling DSE not working?

BidHitter, 2019-03-29:
Hi if I do a system info dump (-i) before disabling the DSE with the -d flag

My CodeIntegrityInformation is as follows:
`SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001`

So then I disable DSE and do a dump again and its as follows:
`SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0000`

Then after running -e to re-enable DSE again and then do a dump IntegrityOptions are still:
`SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0000`

Thanks anyway, great release

BidHitter, 2019-03-29:
Scratch that I didn't see you have to supply the original value when re-enable-ing it. My bad. Nice work again!

closed 2019-03-29 by BidHitter

Mattiwatti, 2019-03-29:
You shouldn't *have* to specify the original value, though if it's not specified a default value of `CODEINTEGRITY_OPTION_ENABLED = 0x1` will be used (correct for your case, but on recent Windows 10 versions the value is usually different). In other words, `EfiDSEFix.exe -d` should have worked for you.

Which OS are you on?

Mattiwatti, 2019-03-30:
I just tested this, and you were too quick to close your issue because it was correct :stuck_out_tongue: 

`EfiDSEFix.exe -e` *should* work without arguments. The reason it's possible to manually specify the `g_CiOptions` value is because I don't actually know the meaning of the values, so the correct value may for example differ per machine. I said that `0x1` would be correct for your machine, but on Windows 8 and higher it should actually be `0x6`. At least that was the case on the machines I tried. I tested and pushed a fix for this.

The problem with these `g_CiOptions` values (or flags?) is that they don't correspond to the output of `NtQuerySystemInformation(SystemCodeIntegrityInformation)`, which is what the `-i` flag dumps, and is usually (but not necessarily) simply `0x0` or `0x1`. On the other hand they do have some overlap, as can be seen from the fact that running `-d` strips the `CODEINTEGRITY_OPTION_ENABLED` bit.

commit d50a2aa7 (2019-03-30): EfiDSEFix: on Windows >= 8, use a different default g_CiOptions value

Fixes #1