Mattiwatti/EfiGuard #2: Is it possible for replacing the existing bootx64.efi by SetFirmwareEnvironmentVariable? :O

Kelvinhack, 2019-04-01:


Mattiwatti, 2019-04-01:
I'm not sure what you mean by this. `SetFirmwareEnvironmentVariableEx` can be used to read from/write to arbitrary kernel mode addresses (via EfiGuardDxe's `SetVariable` hook), but it does not write to or replace existing files, you have to do this yourself.

What exactly is it that you are trying to do?

Kelvinhack, 2019-04-02:
I mean, code it be deploy your Loader without USB stick boot by replacing the Windows's one

Mattiwatti, 2019-04-02:
Ah, I see what you mean now. The answer is **probably yes**, but the way to do this depends somewhat on the target machine, especially whether it is real or virtual, as most VM hosts do not seem to have working persistent NVRAM emulation.

On a real machine or a VM with persistent NVRAM, you *don't* want to overwrite `bootx64.efi`. The reason for this is that during installation, Windows adds a UEFI boot entry named Windows Boot Manager which boots `/EFI/Microsoft/Boot/bootmgfw.efi`. This is the file that is booted by the firmware boot manager. `bootx64.efi` (a safety copy of `bootmgfw.efi` in case the boot entry is lost) is ignored. You also don't want to replace `bootmgfw.efi` with the loader, because the loader will refuse to recursively boot itself. (I've never seen what happens if you do this, I should try it out some day...)

Try the following:
1. In Windows, run `mountvol X: /S` to mount the EFI System Partition at `X:`.
2. Copy `EfiGuardDxe.efi` and `Loader.efi` to `X:\EFI\Boot`, next to `bootx64.efi`.
3. Boot to the UEFI Shell from the BIOS or a USB stick (see README).
4. `cd` to `fsX:\EFI\Boot`, where `X` is the Windows EFI partition.
5. View the current boot options with `bcfg boot dump`.
6. Take `N = last entry+1`, and run `bcfg boot add <N> Loader.efi "EfiGuard"`.
7. (Optional) To set EfiGuard as the default boot option, run `bcfg boot mv <N> 0`

If you have a reasonably compliant BIOS, there should now be a boot menu entry to boot EfiGuard which will find and start Windows. Do not remove the Windows Boot Manager entry, the loader needs this to find the correct `bootmgfw.efi` to boot.

If you **don't** have compliant BIOS, or if `bcfg` commands do not persist due to poor NVRAM emulation (my experience with VMs - VMware could be an exception but I don't use it), it gets trickier. On these types of systems you will need to place the loader at `/EFI/Boot/bootx64.efi` on some volume and make this the default boot option. In this case the loader must be on a different disk (disk, not partition!) from Windows. The reason for this is that while it is theoretically possible to have multiple EFI System Partitions on a non-removable disk, in practice it is poorly supported by both firmware and OS vendors, [including Microsoft](https://support.microsoft.com/en-us/help/2879602/unable-to-boot-if-more-than-one-efi-system-partition-is-present).

Kelvinhack, 2019-04-03:
Thanks for the detailed explanation, Let me check it out

Mattiwatti, 2019-04-03:
Update: I tried this on a Dell XPS and an MSI Z270. It worked on the Dell but not on the MSI. The MSI has an AMI BIOS that seems to get confused by having multiple boot entries for the same disk, because its boot entry granularity stops at 'hard disk', with a separate menu to set the hard disk boot order. This means there is no one-to-one mapping between `bcfg` entries and what is actually booted.

I did manage to set EfiGuard as the first 'hard disk' in the Z270 BIOS, but this resulted in a black screen. I'm not sure if the boot order peculiarity is related to this, but it seems likely since booting EfiGuard from a USB stick works fine. So the conclusion seems to be that whether this will work is dependent on the machine's firmware. There is still the workaround of using a second hard disk, but this is of course more cumbersome.

OkazakiNagisa, 2019-08-11:
Maybe setting Driver#### variable to load EfiGuardDxe driver will work.

Mattiwatti, 2023-03-16:
Closing this as the original question has been answered.

closed 2023-03-16 by Mattiwatti