Mattiwatti/EfiGuard #3: Does EfiGuard work with MBR motherboards?

lolcocks123, 2019-04-17:
*Newbie alert*

I have an MBR motherboard that does not support UEFI. (Gigabyte GA-990FXA-UD3 rev 1.1)

Is it possible to get this to work or does it support UEFI boards only?
I did try to boot it from the USB but I got an operating system not found message.

Mattiwatti, 2019-04-17:
You've got an interesting motherboard there. The Gigabyte page for it claims "Patented DualBIOS with Hybrid EFI technology for 3TB HDD support", which leaves some doubt as to whether or not it is actually a (U)EFI motherboard. If I download the latest BIOS, I can open it in [UEFITool](https://github.com/LongSoft/UEFITool), meaning it has an EFI flash descriptor, but there is no volume top file and the ROM seemingly contains only padding and microcode sections. Note that rev 4.0 of the same motherboard **does** have a full UEFI BIOS. Given the "3TB HDD support" clue I would guess that this is some sort of transitional BIOS which is not actually compliant with any UEFI spec but implements only those features required to recognize and boot from GPT hard disks.

To answer the question: strictly speaking the answer is no, at least not without help. But there are workarounds that may or may not work.

It is possible that your motherboard will recognize that the USB stick is bootable as UEFI if you convert it to a GPT layout. However, since it has such an exotic hybrid BIOS it is uncertain if this boot would actually succeed. If not, [EDK2 Duet](https://github.com/tianocore/tianocore.github.io/wiki/DuetPkg) may work. It is a (now deprecated) framework for running UEFI on a legacy BIOS. It is probably most likely to actually work, but since it is fairly time consuming/error prone to set up, I would try the following (in order of time needed):
1. Convert your USB stick to GPT. Windows `diskpart` or `diskmgmt.msc` cannot do this, but most third party partition management software can. I use [MiniTool Partition Wizard](https://www.partitionwizard.com). See if the BIOS can boot from the stick. If it still can't, try step 2. If it tries to boot but fails (a black screen or hang counts), go to step 3.
2. Try installing EfiGuard on your Windows EFI partition instead of a USB stick. See #2 for information on how to do this.
3. Boot Duet, and use Duet to boot EfiGuard (...to boot Windows). I have never used Duet myself so I can't be of much help here. I will say though that it is probably best to ignore the official repository, since the readme is aimed at UEFI developers and not very beginner friendly. Instead you can use a prepackaged version and modify it if needed. [This thread](https://www.win-raid.com/t3286f50-Guide-NVMe-boot-for-systems-with-legacy-BIOS-and-older-UEFI-DUET-REFIND.html) is an example of using such a distribution to boot NVMe disks on non-UEFI motherboards through Duet.

One thing to consider is that your Windows installation **must** be an EFI one for EfiGuard to work, regardless of anything else. There are guides on how to convert an existing installation to EFI if you search for this, but again this presumes a UEFI compatible BIOS. If your BIOS cannot boot regular Windows in EFI mode, it cannot boot EfiGuard. In this case Duet is your only option.

lolcocks123, 2019-04-21:
Thank you for the quick response and I apologize for replying so late.

Duet doesn't seem to be working for my motherboard and the lack of support for it has my hopes down.

Can something like [Clover EFI bootloader](https://sourceforge.net/projects/cloverefiboot/) work in my case?

Mattiwatti, 2019-04-21:
Clover might work, I don't know. If you find a tutorial/guide on how to use it as an MBR bootloader to chain to UEFI, please link it so I can take a look. As far as I know Clover is mainly used to boot Mac OS on Hackintosh PCs, but that doesn't mean it can't work for your case.

I tried Clover and Duet on an Asus P7P55D. Clover failed the same way as trying to boot a GPT disk would have. Duet at least booted, but failed with "error: MBR start!" which is not exactly a very helpful error message (for one, it doesn't say which disk is the one at fault, and two, converting the USB stick from MBR to GPT results in the same failure as Clover). The error is weird in any case since I'm certain Duet is meant to be booted from MBR media, since that is the whole reason it exists in the first place. Alas, grepping the source for the error message didn't turn up anything (nor did Google), so I'm at a loss there.

I'm probably using Clover wrong, but in the absence of any documentation that I could see I simply created a bootable (MBR) USB stick from the ISO using Rufus.

Mattiwatti, 2019-04-21:
Answered my own question: using [this tool](https://www.aioboot.com/en/clover-bootloader-windows/) I was able to create a USB stick that can boot a working UEFI shell with Clover on the P7P55D. Warning: the tool is not open source and I don't really feel like spending the time needed to reverse it right now. The tool also only supports Windows. Presumably on a Mac you can just run the installer, but who uses Macs?

lolcocks123, 2019-04-21:
Yep, Clover Legacy Installer is exactly what I tried, unfortunately, I just ended up with a black screen with the cursor on top.
While Duet just has me stuck here:
https://i.ibb.co/DkRd2Fh/f50t3286p75195n2-t-FYUbr-PQ.jpg

I guess I am out of luck.

To shed more light on why I wanted EfiGuard, is [DifferentSLIAuto](https://www.techpowerup.com/forums/threads/sli-with-different-cards.158907/page-129#post-4027277)
Using two cards needs the drivers to be modified, causing them to become unsigned causing conflicts with anti cheats.

Using your tool, one guy ([DonnerPartyOf1](https://www.techpowerup.com/forums/members/donnerpartyof1.186546/)) was able to get Apex Legends working through a VM, but I guess it won't work directly without a VM as I would still have a unsigned driver running in the kernel space which anti cheats hate. I guess a VM is cloaking that fact.

Anyway, thank you for all your help Mattiwatti, really appreciate it. 
Thanks for this tool, I will surely use it after a get a new motherboard. :)

Mattiwatti, 2019-04-21:
I see, that's a shame. I guess your motherboard really doesn't want to boot anything UEFI. It's always hit and miss with these things depending on the hardware. Will close this issue.

Re: VM vs non-VM: EfiGuard should work with either. It is disabling DSE that is allowing the unsigned driver to run, not running in a VM.

closed 2019-04-21 by Mattiwatti

acoul, 2023-06-01:
[This thread](https://www.win-raid.com/t3286f50-Guide-NVMe-boot-for-systems-with-legacy-BIOS-and-older-UEFI-DUET-REFIND.html) is an example of using such a distribution to boot NVMe disks on non-UEFI motherboards through Duet.

just a [broken URL fix](https://winraid.level1techs.com/t/guide-nvme-boot-for-systems-with-legacy-bios-and-uefi-board-duet-refind/32251)