Mattiwatti/EfiGuard #4: error C0000002

Rikudouu, 2019-06-21:
I'm on Windows 1809 (soon 1903), and I get this error when trying any command other than -i

```
Failure: NtSetSystemEnvironmentValueEx error C0000002
AdjustCiOptions failed: C0000002
```

Full error with -d:
```
Disabling DSE...
CI!g_CiOptions at 0xFFFFF80514541D18.
Failure: NtSetSystemEnvironmentValueEx error C0000002
AdjustCiOptions failed: C0000002
```

I also pulled from master and compiled the EfiDSEFix but the same issue occurs.

Mattiwatti, 2019-06-21:
> - FirmwareType: BIOS

That would be the problem. EfiGuard is a UEFI bootkit, it does not work with BIOS installations.

If you have a UEFI-capable motherboard, you need to either reinstall Windows in UEFI mode or convert your existing installation to UEFI (there are guides on how to do this which you can find on Google).
If you have a non-UEFI motherboard, you can try some of the workarounds listed in #3, which may or may not work depending on the motherboard.

Rikudouu, 2019-06-21:
Oh, I'm an idiot. It's been a while since I installed Windows on this pc. Converting my boot drive to GPT fixed it, thanks.

closed 2019-06-21 by Rikudouu

Hiei-YYH, 2023-03-28:
having the same problem here, my system volume says "GUID (GPT)" in which windows 11 is installed... but i receive the same error, (it was legacy/uefi before, then i changed to uefi only) does it mean i need to reinstall windows?

```
Disabling DSE...
CI!g_CiOptions at 0xFFFFF804628AD004.
Failure: NtSetSystemEnvironmentValueEx error C000000D
AdjustCiOptions failed: C000000D
```

Mattiwatti, 2023-03-28:
That is not the same problem since you are seeing a different status code. Furthermore, this issue was closed, please refrain from posting in it unless you believe it should be reopened. Furtherfurthermore, you seem to be running an older version of EfiDSEFix. The latest release won't fix your problem but it may point you in the right direction better.

Please refer to (and read...) #60 and then post there if you are still seeing this problem after that.