Mattiwatti/EfiGuard #5: EfiGuardUnload causes infinite bootx64.efi booting loop

diejozef, 2019-06-25:
Title says it all.
After EfiGuardUnload is called, it will never try to boot to bootmgfw.

Mattiwatti, 2019-06-28:
Hi,

Can you elaborate please? Are you saying that the EfiGuard loader is recursively booting itself due to the driver somehow calling `EfiGuardUnload` on itself instead of patching `bootmgfw.efi`? Or am I misunderstanding?

Also, some details please:
- What are the contents of `bootx64.efi` in this scenario? Is it the EfiGuard loader or the original loader of some OS?
- Are you trying to boot Windows, or are you trying to boot another OS despite EfiGuard being loaded (which should be supported)?
- What is your UEFI boot configuration (the output of `bcfg boot dump` in the UEFI shell would be great), and are you booting EfiGuard automatically via the boot configuration, or overriding the default boot option with e.g. F10?
- Is this a physical machine or a VM? If the former, I'd like to know the specs, especially the motherboard model and BIOS version. If the latter, please upload the VM configuration (`.vbox`/`.vmx` file contents, QEMU command line...) somewhere if it's not too much trouble.

diejozef, 2019-06-28:
Yes, the problem is exactly what you described, seems as if the first image that gets passed into the LoadImage hook was bootx64 and after unloading it tries to load itself recursively.

`bootx64.efi` is the `EfiGuard` loader and I'm using Windows 10 build 1809. 
This is a physical machine, mobo is  an Asus ROG Maximus XI Extreme (chipset Z390) with BIOS version `0224 (American Megatrends)`.
I set the primary boot device to a FAT32 formatted flash drive in BIOS so it boots automatically from there. I'll try to dump my cfg from UEFI shell when I get back to my PC if it's still needed.

Mattiwatti, 2019-06-28:
Yes please, the output of the following two would be helpful:
- (in UEFI Shell) `bcfg boot dump`
- (in Windows) `bcdedit /enum all`

Note: you can redirect stdout in the UEFI shell just like on Windows/Linux, so you can just save your output with `bcfg boot dump > bcfg.txt`.

Mattiwatti, 2019-06-29:
And another request... I've searched for BIOS versions for the ROG Maximus XI variants, and your version `0224` is actually a pre-release version (it can't be found on the Asus site for one). `0506`, dated 2018-10-31, was the official release BIOS, and `1105`, dated 2019-06-21, is the current most up to date BIOS.

Could you please update to the `1105` version? For most motherboards this does unfortunately mean you will have to reconfigure the BIOS after upgrading. On the plus side, there have been a number of fairly important updates to this BIOS since its release, so I would recommend upgrading anyway. The BIOS files are available for download [here](https://www.asus.com/us/Motherboards/ROG-MAXIMUS-XI-EXTREME/HelpDesk_BIOS/). You do not need to install the separate Intel ME update because the `1105` image already contains the latest version of the ME.

The reason I'm asking you to do this is because
1. It is possible that the boot loop you are seeing is the result of some obscure bug in the motherboard's  BDS phase that has been fixed in a newer version. I would say this is unlikely, but I wouldn't completely rule it out either.
2. I need to have the same BIOS version as the one installed on your machine in order to do a useful static analysis. It is theoretically possible for you to make a dump of your current BIOS with the Intel Flash Programming Tool, but this is both more work than simply upgrading and also leaves you without the performance/stability/compatibility improvements from the upgrade.

diejozef, 2019-06-29:
Sorry for the delay, here is the output of `bcfg boot dump`:
```
Option: 00. Variable: Boot0001   
  Desc    - UEFI: Sony Storage Media PMAP, Partition 1
  DevPath - PciRoot(0x0)/Pci(0x14,0x0)/USB(0x17,0x0)/HD(1,MBR,0xE7175B12,0x800,0x1CF1800)
  Optional- Y
Option: 01. Variable: Boot0000   
  Desc    - Windows Boot Manager
  DevPath - HD(2,GPT,D6DB5935-FC71-47A8-B325-CEFBA55EDC21,0xFA000,0x31800)/\EFI\MICROSOFT\BOOT\BOOTMGFW.EFI
  Optional- Y
```

and the output of `bcdedit /enum all`:
```

Firmware Boot Manager
---------------------
identifier              {fwbootmgr}
displayorder            {03671e21-96e2-11e9-a553-806e6f6e6963}
                        {bootmgr}
timeout                 1

Windows Boot Manager
--------------------
identifier              {bootmgr}
device                  partition=\Device\HarddiskVolume4
path                    \EFI\MICROSOFT\BOOT\BOOTMGFW.EFI
description             Windows Boot Manager
locale                  en-US
inherit                 {globalsettings}
default                 {current}
resumeobject            {41c4d32b-2c67-11e9-9445-afcf21eec875}
displayorder            {current}
toolsdisplayorder       {memdiag}
timeout                 30

Firmware Application (101fffff)
-------------------------------
identifier              {03671e21-96e2-11e9-a553-806e6f6e6963}
device                  partition=E:
description             UEFI: Sony Storage Media PMAP, Partition 1

Windows Boot Loader
-------------------
identifier              {current}
device                  partition=C:
path                    \Windows\system32\winload.efi
description             Windows 10
locale                  en-US
inherit                 {bootloadersettings}
recoverysequence        {41c4d32d-2c67-11e9-9445-afcf21eec875}
displaymessageoverride  Recovery
recoveryenabled         Yes
isolatedcontext         Yes
allowedinmemorysettings 0x15000075
osdevice                partition=C:
systemroot              \Windows
resumeobject            {41c4d32b-2c67-11e9-9445-afcf21eec875}
nx                      OptIn
bootmenupolicy          Standard

Windows Boot Loader
-------------------
identifier              {41c4d32d-2c67-11e9-9445-afcf21eec875}
device                  ramdisk=[\Device\HarddiskVolume3]\Recovery\WindowsRE\Winre.wim,{41c4d32e-2c67-11e9-9445-afcf21eec875}
path                    \windows\system32\winload.efi
description             Windows Recovery Environment
locale                  en-us
inherit                 {bootloadersettings}
displaymessage          Recovery
osdevice                ramdisk=[\Device\HarddiskVolume3]\Recovery\WindowsRE\Winre.wim,{41c4d32e-2c67-11e9-9445-afcf21eec875}
systemroot              \windows
nx                      OptIn
bootmenupolicy          Standard
winpe                   Yes

Resume from Hibernate
---------------------
identifier              {41c4d32b-2c67-11e9-9445-afcf21eec875}
device                  partition=C:
path                    \Windows\system32\winresume.efi
description             Windows Resume Application
locale                  en-US
inherit                 {resumeloadersettings}
recoverysequence        {41c4d32d-2c67-11e9-9445-afcf21eec875}
recoveryenabled         Yes
isolatedcontext         Yes
allowedinmemorysettings 0x15000075
filedevice              partition=C:
filepath                \hiberfil.sys
bootmenupolicy          Standard
debugoptionenabled      No

Windows Memory Tester
---------------------
identifier              {memdiag}
device                  partition=\Device\HarddiskVolume4
path                    \EFI\Microsoft\Boot\memtest.efi
description             Windows Memory Diagnostic
locale                  en-US
inherit                 {globalsettings}
badmemoryaccess         Yes

EMS Settings
------------
identifier              {emssettings}
bootems                 No

Debugger Settings
-----------------
identifier              {dbgsettings}
debugtype               Local

RAM Defects
-----------
identifier              {badmemory}

Global Settings
---------------
identifier              {globalsettings}
inherit                 {dbgsettings}
                        {emssettings}
                        {badmemory}

Boot Loader Settings
--------------------
identifier              {bootloadersettings}
inherit                 {globalsettings}
                        {hypervisorsettings}

Hypervisor Settings
-------------------
identifier              {hypervisorsettings}
hypervisordebugtype     Serial
hypervisordebugport     1
hypervisorbaudrate      115200

Resume Loader Settings
----------------------
identifier              {resumeloadersettings}
inherit                 {globalsettings}

Device options
--------------
identifier              {41c4d32e-2c67-11e9-9445-afcf21eec875}
description             Windows Recovery
ramdisksdidevice        partition=\Device\HarddiskVolume3
ramdisksdipath          \Recovery\WindowsRE\boot.sdi
```

I've updated my BIOS to the latest one (version 1105), uncommented [this](https://github.com/Mattiwatti/EfiGuard/blob/master/EfiGuardDxe/EfiGuardDxe.c#L220) line and compiled the driver again, same looping bug happens. The only output I see (before it starts booting itself over and over again) is from [here](https://github.com/Mattiwatti/EfiGuard/blob/master/EfiGuardDxe/EfiGuardDxe.c#L159) (aside from everything in `EfiGuardInitialize`)

Mattiwatti, 2019-07-09:
Thanks for your patience - it actually took me a LOT of time to figure out this one :) Even getting it to reproduce took me quite a while. The fix is of course embarrassingly simple and obvious in hindsight. Ah well....

Can you please confirm that the latest commit fixes the issue for you? Then I will put out a new release.

closed 2019-07-09 by Mattiwatti

diejozef, 2019-07-17:
Yes, it is working properly now.
Thank you for the insane amount of time you put into resolving this issue.

voraz415, 2019-08-30:
Can you put out the new release?  I am having the same problem using latest release on github

Mattiwatti, 2019-10-07:
Apologies for the delay, I forgot about the notification I got for this.

I've just published a new release. The zip file for v1.0.2 should include this fix.

commit f9270cfb (2019-07-09): Fix potential recursive self-boot by Loader.efi

The EFI variable "BootCurrent" corresponds to XXXX in some variable "BootXXXX", and this value was being used to guard against recursive self-booting. However, EfiBootManagerGetLoadOptions() already returns the options sorted by the contents of the "BootOrder" variable, which means that XXXX does not necessarily correspond to the linear index in a for loop. Instead use BootOptions[Index].OptionNumber to retrieve and compare the original value of XXXX

Fixes #5