Mattiwatti/EfiGuard #7: DSEFix on boot without config tool?

godmodeGoat, 2019-10-05:
how can i set DSEFix to activate on boot, without having to use the config tool manually each time?

I know i can create a task to run efidsefix.exe on logon (or even pre-logon) but i'd like to know if there is a way to integrate it into the boot process?

Mattiwatti, 2019-10-05:
Hi,

Can you elaborate what you mean by 'activate on boot' exactly? Do you mean that you want EfiGuardDxe (meaning the UEFI driver, not a Windows application) to flip the `g_CiOptions`/`g_CiEnabled` bit at some point instead of `EfiDSEFix.exe`? If so, that isn't really possible. Why? Consider this timeline:
- EfiGuardDxe is loaded.
- `bootmgfw.efi` is loaded.
- `winload.efi` is loaded.
- **UEFI ExitBootServices is called**. EfiGuardDxe uses this as a last chance to display patch information to the user.
- After this point, EfiGuardDxe can no longer act 'of itself' and can only be called into via the EFI runtime services (this is what `EfiDSEFix.exe` does).
- `SepInitializeCodeIntegrity` is called by the kernel and `gi_CiOptions`/`gi_CiEnabled` is set to the 'enabled' value by `CI.dll`.

As you can see, this does not leave any room for EfiGuardDxe to disable DSE after it has actually been enabled for the first time.

The alternative, loading the driver with the 'disable DSE at boot' option specified (using `Loader.config.efi`) **does** disable DSE, but does so permanently (at least until reboot) and detectably. Since this is in the README I'm guessing you have already considered and dismissed this option for your use case. If this **is** an option for you, I would recommend compiling EfiGuard yourself (or at least `Loader.efi`) so that you do not get the prompt at boot every time. To do this, simply modify `EfiGuardDxe.c` [here](https://github.com/Mattiwatti/EfiGuard/blob/f9270cfb3897ec68f36c5d032127aa59587356bd/EfiGuardDxe/EfiGuardDxe.c#L44) and change `DSE_DISABLE_SETVARIABLE_HOOK` to `DSE_DISABLE_AT_BOOT`.

sakurawill, 2022-02-16:
> Hi,
>
> Can you elaborate what you mean by 'activate on boot' exactly? Do you mean that you want EfiGuardDxe (meaning the UEFI driver, not a Windows application) to flip the `g_CiOptions`/`g_CiEnabled` [...]

hello,sorry to bother you. I've tried to compile EfiGuard myselft. but i faild to do that. could you help me to compile one with DSE_DISABLE_AT_BOOT. thanks a lot.

closed 2023-03-16 by Mattiwatti