Mattiwatti/EfiGuard #11: Issue when booting

SoftCrush, 2020-04-19:
I can't seem to be able to boot using either the loader or the UEFI shell.

When using the loader, I select the USB and I get an error saying "The operating system wasn't found. Try disconnecting any drives that don't contain an operating system. Press any key to restart". After pressing a key this eventually leads to an error saying "No boot disk has been detected or the disk has failed."

When using the UEFI shell to load the driver, I get to the point where the title message is shown (Rootkits You Can Trust). However, my screen turns soon black with a blinking cursor and I have to turn off my computer.

Any ideas?

Mattiwatti, 2020-04-24:
What motherboard do you have? And with what BIOS version?
Also, can you post the output of `EfiDSEFix.exe -i` (run from Windows) here?

devastor20, 2023-05-13:
i have same issue here.
stack after load rootkit, with blank monitor and my pc still stand by.
i will share my EfiDSEFix.exe -i 

SystemBootEnvironmentInformation:
        - BootIdentifier: {83aef3ea-f16e-11ed-8001-6d7368757261}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001
           0x0001: CODEINTEGRITY_OPTION_ENABLED

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000

SystemIsolatedUserModeInformation:
        - SecureKernelRunning: 0
        - HvciEnabled: 0
        - HvciStrictMode: 0
        - DebugEnabled: 0
        - FirmwarePageProtection: 0
        - EncryptionKeyAvailable: 0
        - TrustletRunning: 0
        - HvciDisableAllowed: 0

Mattiwatti, 2023-10-03:
@devastor20 Can you run `load -nc EfiGuardDxe.efi` (so adding "-nc") from the UEFI Shell and report if this fixes the issue? (Note: you cannot use `Loader.efi` for this, so you will need to run `bootmgfw.efi` manually from the UEFI Shell afterwards to start Windows.)

Same question for @SoftCrush actually - this is a bit of a late brainwave on my part but hey, better later than never...

If this does **not** work, I am back to having no clue.

devastor20, 2023-10-04:
hi @Mattiwatti thank you for your reply, I appreciate it.
I don't remember what I did till it worked, but I reinstalled my Windows and it works perfectly.
well, it's not like I can hack any games after that because I just started but at least I can use CE fine and without detected.
I really love your work. Thank you

Mattiwatti, 2023-10-04:
Absent a response from @SoftCrush I'm going to assume this is either fixed or no longer relevant, but feel free to reopen if not.