Mattiwatti/EfiGuard #14: EfiDSEFix Supported versions

v0idi, 2020-05-29:
Can i use EfiDSEFix  on anything past 1703 ?

Mattiwatti, 2020-05-29:
Sure. It should work on any EFI-compatible version of Windows, including all versions of Windows 10 released to date.

v0idi, 2020-05-29:
Thank you so much for the fast reply
im getting a weird freeze after booting from the usb i just get a black screen , does it wait for an input or something? i can boot the loader fine but with the EfiGuardDxe.efi driver i get that problem.

Here is EfiDSEFix.exe -i output :

SystemBootEnvironmentInformation:
        - BootIdentifier: {195d91a2-40cd-11ea-ba77-8ec75fbd906c}
        - FirmwareType: BIOS
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000
im on 1903 btw

v0idi, 2020-05-29:
also im using the latest release EfiGuard-v1.1

Mattiwatti, 2020-05-29:
`- FirmwareType: BIOS`

This is your problem. Windows must be running in UEFI mode for EfiGuard to work. You can either reinstall Windows, or [convert your installation](https://www.windowscentral.com/how-convert-mbr-disk-gpt-move-bios-uefi-windows-10) (NB: this is just the first result I found on Google, I haven't actually tested this method).

Note that you do need a motherboard capable of booting from a GPT disk (i.e. with a UEFI BIOS). This is almost certainly the case nowadays, but you should still check this if you aren't sure.

v0idi, 2020-05-29:
Thank you that was the case although my boot drive was already on gpt but i used mbr2gpt anyway.
now when loading the driver , after patching winload!OsFwpKernelSetupPhase1 
windows tries to boot but it get stuck at the windows logo (in this case my mobo manufacturers) 
after a few tries it triggers the automatic repair loop

Here is EfiDSEFix.exe -i output :
SystemBootEnvironmentInformation:
        - BootIdentifier: {b301a22f-a212-11ea-b8b2-b4bbd2714ec1}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000

v0idi, 2020-05-29:
nevermind lol i had an efi driver on that usb that was causing the problem 
Disabling DSE...
CI!g_CiOptions at 0xFFFFF804589C7278.
Successfully disabled DSE. Original g_CiOptions value: 0x16
Thank you so much for your help !

closed 2020-05-29 by v0idi