Mattiwatti/EfiGuard #15: NMAKE : fatal error U1077: "GenFw:

stonedreamforest, 2020-08-18:
```
E:\github\edk2>build -a X64 -t VS2019 -p EfiGuardPkg/EfiGuardPkg.dsc -b RELEASE
Build environment: Windows-10-10.0.18362-SP0
Build start time: 17:25:22, Aug.18 2020

WORKSPACE        = e:\github\edk2
EDK_TOOLS_PATH   = e:\github\edk2\basetools
EDK_TOOLS_BIN    = e:\github\edk2\basetools\bin\win32
CONF_PATH        = e:\github\edk2\conf
PYTHON_COMMAND   = py -3

Processing meta-data .

Architecture(s)  = X64
Build target     = RELEASE
Toolchain        = VS2019

Active Platform          = e:\github\edk2\EfiGuardPkg\EfiGuardPkg.dsc
. done!
Building ... e:\github\edk2\MdePkg\Library\BaseSynchronizationLib\BaseSynchronizationLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiLib\UefiLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiDriverEntryPoint\UefiDriverEntryPoint.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BaseTimerLibNullTemplate\BaseTimerLibNullTemplate.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiRuntimeServicesTableLib\UefiRuntimeServicesTableLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiDevicePathLibDevicePathProtocol\UefiDevicePathLibDevicePathProtocol.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BasePrintLib\BasePrintLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiMemoryAllocationLib\UefiMemoryAllocationLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiBootServicesTableLib\UefiBootServicesTableLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BaseDebugLibNull\BaseDebugLibNull.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BaseMemoryLibOptDxe\BaseMemoryLibOptDxe.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BaseLib\BaseLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\DxePcdLib\DxePcdLib.inf [X64]
Building ... e:\github\edk2\MdeModulePkg\Library\UefiBootManagerLib\UefiBootManagerLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\UefiApplicationEntryPoint\UefiApplicationEntryPoint.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BasePerformanceLibNull\BasePerformanceLibNull.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\BasePeCoffGetEntryPointLib\BasePeCoffGetEntryPointLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\DxeServicesTableLib\DxeServicesTableLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\DxeServicesLib\DxeServicesLib.inf [X64]
Building ... e:\github\edk2\MdeModulePkg\Library\DxeReportStatusCodeLib\DxeReportStatusCodeLib.inf [X64]
Building ... e:\github\edk2\MdeModulePkg\Library\UefiHiiLib\UefiHiiLib.inf [X64]
Building ... e:\github\edk2\MdeModulePkg\Library\UefiSortLib\UefiSortLib.inf [X64]
Building ... e:\github\edk2\MdePkg\Library\DxeHobLib\DxeHobLib.inf [X64]
Building ... e:\github\edk2\MdeModulePkg\Library\UefiHiiServicesLib\UefiHiiServicesLib.inf [X64]
Building ... e:\github\edk2\EfiGuardPkg\Application\Loader\Loader.inf [X64]
Building ... e:\github\edk2\EfiGuardPkg\EfiGuardDxe\EfiGuardDxe.inf [X64]
        "GenFw" -e UEFI_APPLICATION -o e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\Application\Loader\Loader\OUTPUT\Loader.efi e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\Application\Loader\Loader\DEBUG\Loader.dll --keepexceptiontable --keepzeropending --keepoptionalheader
        "GenFw" -e DXE_RUNTIME_DRIVER -o e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\EfiGuardDxe\EfiGuardDxe\OUTPUT\EfiGuardDxe.efi e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\EfiGuardDxe\EfiGuardDxe\DEBUG\EfiGuardDxe.dll --keepexceptiontable --keepzeropending --keepoptionalheader
'GenFw' ڲⲿҲǿеĳ
ļ
NMAKE : fatal error U1077: "GenFw: ش롰0x1
Stop.

build.py...
 : error 7000: Failed to execute command
        C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.27.29110\bin\Hostx86\x86\nmake.exe /nologo tbuild [e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\Application\Loader\Loader]

build.py...
 : error 7000: Failed to execute command
        C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.27.29110\bin\Hostx86\x86\nmake.exe /nologo tbuild [e:\github\edk2\Build\EfiGuard\RELEASE_VS2019\X64\EfiGuardPkg\EfiGuardDxe\EfiGuardDxe]

build.py...
 : error F002: Failed to build module
        e:\github\edk2\EfiGuardPkg\Application\Loader\Loader.inf [X64, VS2019, RELEASE]

- Failed -
Build end time: 17:25:24, Aug.18 2020
Build total time: 00:00:02

```

hello, how to fix this error?

Mattiwatti, 2020-08-19:
I can't reproduce this here, using the latest EDK2 master (@ 5a6d764).

Which version of EDK2 are you using? And did you compile the base tools yourself or are you using the precompiled Win32 package?

stonedreamforest, 2020-08-20:
I re-clone the latest commits and compile everything is normal, maybe I missed something before😄.

closed 2020-08-20 by stonedreamforest