Mattiwatti/EfiGuard #18: calling KeGetCurrentIrql in SetVariable hook

diejozef, 2020-08-25:
hi,

I tried calling `KeGetCurrentIrql` (after booting into Windows, so everything should've been loaded by then) in the `SetVariable` hook but it returned a strange value (decimal 125). Acquired function pointer via `GetProcedureAddress`.

Are there limitations of what kernel API can be called from this backdoor that I don't know of?

thank you in advance!

Mattiwatti, 2020-08-25:
Is your goal really to call `KeGetCurrentIrql()` or are you using this as a stepping stone to calling further Windows API functions? `KeGetCurrentIrql()` on x64 is nothing more than a read of the CR8 register, so I would just do that using the `__readcr8` intrinsic. Nevertheless, calling the kernel's `KeGetCurrentIrql()` export should work. Here's some sample code that works for me:
```
// ...snip...
case 8:
{
    typedef
    UINT8
    (EFIAPI*
    t_KeGetCurrentIrql)(
        VOID
        );

    PEFI_IMAGE_NT_HEADERS NtHeaders = RtlpImageNtHeaderEx(gKernelPatchInfo.KernelBase, 0);
    t_KeGetCurrentIrql KeGetCurrentIrql = (t_KeGetCurrentIrql)GetProcedureAddress((UINTN)gKernelPatchInfo.KernelBase, NtHeaders, "KeGetCurrentIrql");
    BackdoorData->u.Qword = (UINT64)KeGetCurrentIrql();
    break;
}
```
(I returned an 8 byte/qword value here, but the 1 byte field would have also worked due to the return type of `KeGetCurrentIrql()`. Note that the `__readcr8()` intrinsic **does** return a 64 bit value, although in practice it will not exceed `sizeof(unsigned char)` on Windows.)

> Are there limitations of what kernel API can be called from this backdoor that I don't know of?

The `SetVariable` function will be called with a certain system affinity mask set by the HAL for the duration of the call, which I have to say I've never investigated. It may be something as simple as constraining the thread to a single logical or physical core. If this is an issue you can always call `KeSetSystemGroupAffinityThread` again and revert back with `KeRevertToUserGroupAffinityThread` when done.

Other things to be aware of about the environment `SetVariable` is executed in:
- The code is running in a critical region (i.e. kernel APCs are disabled).
- The global fast mutex `ExpEnvironmentLock` is held. This is only an issue if you attempt to call other EFI environment variable related functions in the kernel (e.g. `NtSetBootOptions`). Doing so will cause a deadlock.

diejozef, 2020-08-25:
Thank you for your help, I realized I didn't pass the correct nt headers. also extra thanks for providing some info about the environment the hook is running in.

closed 2020-08-25 by diejozef