Mattiwatti/EfiGuard #19: Your thoughts on 2021 driver certificates

frostiest, 2020-08-26:
Hi Matti,
as you likely know in mid 2021 Microsoft will be forcing all drivers to be submitted, approved, and signed by them, each and every time they want to sign ...3rd party sellers will no longer be permitted to sell certificates and any existing drivers will be rendered useless once their certificates expire. This change will be pushed to all windows even windows xp.

my question is do you think efiguard will be affected? even though it's a boot driver? It's my understanding a lot of machines don't support secure boot so not reasonable to try and enforce that? or what are your thoughts.

closed 2020-08-27 by frostiest

Mattiwatti, 2020-08-29:
Do you have any source for your claims? I haven't been able to find any information confirming this.

Regardless of whether this is true or not, EfiGuard is not affected as it is not a boot driver (i.e. a driver that depends on the Windows kernel), but a bootkit, and as such does not need to follow MS kernel signing rules. However, it **is** (and has always been) subject to any Secure Boot policy present on the machine.

frostiest, 2020-08-30:
> Do you have any source for your claims? I haven't been able to find any information confirming this.
>
> Regardless of whether this is true or not, EfiGuard is not affected as it is not a boot driver [...]

Hi,
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates

and here will be the registering step before having to submit
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program

here's 3rd party sellers mentioning the coming changes at the top
https://www.digicert.com/code-signing/kernel-mode-certificates.htm

Mattiwatti, 2020-08-31:
I see. What is happening is that Microsoft will be removing support for cross-signed root certificates that have kernel mode signing capabilities. In practice I expect that this will not change much for people who are already paying for certificates, as EV certificates have been the recommended way to do kernel mode signing for Windows 10 for a long time now. The associated hardware dev center that you have to sign up for is not new either.

It annoys me somewhat that this change will break kernel mode signing certificates for XP in the future for no good reason, but then again it's not like Windows XP strongly resists attempts to install unsigned drivers (unlike Windows 10). You'll get a warning message at worst. Furthermore, so long as SHA1 signing will be available on the dev center for backwards compatibility, I suspect there will still be workarounds possible to make signed driver packages for XP, regardless of what MS says.

As for my opinion about driver signing _in general_ (i.e. not this specific change, which I believe is fairly minor), I can be brief. I believe it is an anti-feature and a way for Microsoft to control what is allowed to run on a device that is mine and not theirs; in other words, they are overstepping their bounds. My strong opposition to driver signature enforcement in Windows is one of the main reasons why I wrote EfiGuard.