Mattiwatti/EfiGuard #20: Issue when booting.

DrogonRocks, 2020-08-31:
Hello, im writting here to seek help, this is my last resort after trying almost everything and having 0 success on disabling patchguard
im currently using an Alienware 15R3 laptop, running W10 19041.450 build, im facing the following issues (not blaming your tool, instead it's my fault for being such a noob) if i try method 1 (Booting the loader) it only "works" if i have secure boot off, i rename Loader.efi  to bootx64.efi but when it boots it just stays on the screen that says matti rootkits you can trust, shows like 2 lines of green text and that's it, it stays like that forever so i have to turn it off manually, however if i try the shell method it "works" until some point, on setp 3 you say that i have to cd to /efi/boot but when i type cd and /efi/boot it says "unkown flag /efi/boot" i don't know if im typing something wrong or what, if i enter "load EfiGuardDxe.efi" it says: file not found, but when i enter either loader.efi or loader.config.efi i get redirected to the screen that says matti rootkits you can trust and it shows lots of green lines, at the end it says "successfully disabled patchguard, Successfully patched ntoskrnl.exe" but i think it stills enabled somehow, it should create a patched entry right? also, if i try to run on desktop EfiDSEFix.exe nothing happens, the file closes itself (AV turned off) also, i think it still disabled because when im using WKE to hide the process i need i keep getting bsod after some minutes, that's why im trying to disable patchguard, hope you can help me or guide me.

Mattiwatti, 2020-09-02:
I edited your post a bit for readability. I don't normally edit people's posts, but you essentially posted one long run-on sentence that was impossible to parse. Please use paragraphs next time...

You are posting about a few different things - I'll start with the first one since it's the simplest to answer. EfiGuard does not work with Secure Boot by default. It is possible to sign EfiGuard with a key that you have added to your firmware's `db` keystore to make it work with Secure Boot, but doing this is nontrivial. See [Sakaki's EFI Install Guide: configuring Secure Boot](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Configuring_Secure_Boot) if you really want to do this. My advice would be to simply disable Secure Boot unless you really need it for some reason.

Re: the loader hanging and making it so you have to turn off the PC: this should obviously not be happening. Unfortunately, without having your exact type of laptop (especially the motherboard) it is very hard for me to fix issues like this since being able to reproduce the problem is a prerequisite for being able to fix it. I have never encountered such a hang on any machine (physical or virtual), and I have used EfiGuard on quite a few devices by now, including some obscure laptop brands with questionable UEFI implementations. My only guess is that your motherboard's BIOS is somehow at fault, so you could see if there is an update available that fixes this.

>  but when i type cd and /efi/boot it says "unkown flag /efi/boot"

`cd` needs to be on a filesystem device before it can work. These are called `fs0`, `fs1`, `fs2`, ... in the UEFI shell, and they represent the available partitions that are mapped. Think of them as `C:` or `D:` on Windows. You switch to a filesystem device by typing its name followed by a colon, e.g.:
`fs0:`
It can sometimes be difficult to tell which filesystem is the one you need, so you can use `ls` to help with this. The EFI partition will always contain a folder called `EFI` (and usually only that), so it is easy to recognise. After you are on the correct filesystem, `cd` to `EFI/boot` as per the `README`.

> When i enter either loader.efi or loader.config.efi i get redirected to the screen that says matti rootkits you can trust and it shows lots of green lines, at the end it says "successfully disabled [...]

This sounds completely normal and is the expected behaviour. What do you mean by "patched entry"? What kind of entry should have been created where?

> i think it still disabled because when im using WKE to hide the process i need i keep getting bsod after some minutes, that's why im trying to disable patchguard

Are you sure this BSOD is caused by Patchguard (bugcode `0x109`) and not by something else? Hiding processes (in Windows, but really any operating system) from the kernel that is supposed to be able to manage these processes can lead to various issues of all sorts, not all of which are related do PatchGuard.

DrogonRocks, 2020-09-04:
Hi, first of all im really sorry for writing that mess, English is not my main language and when i wrote that i was not thinking straight because i was a bit frustrated because i didn't know what i was doing wrong (still don't know but i will try many things and update my post after i try).
 I have in mind do a full bios reset, make a clean windows install and try turning off HyperGuard since you say that it could create problems with Efiguard, secure boot will remain disabled and after that i will boot EfiGuard and try the easiest method and see how it goes now

About "patched entry" i was referring to other methods to disable PatchGuard in older versions of Windows 10, for example after running the file it creates a patched boot entry to select from when starting the computer for example: you can select Windows 10 or Windows 10 Patched, that's why i was asking that, since im a noob and it's been a long time since i disabled patch guard i was expecting that boot menu with a patched entry

Another quick question, do i need to disable something else on BIOS aside from secure boot? do i need to disable Virtualization? i don't need it, im just asking since almost all options are enabled by default and i have them like that.

About my motherboard here's some more detailed info i got from a program: 
Model: Alienware 15 R3  
Manufacturer: Alienware 
Processor revision:  A00  
Serial number: .HTRFRF2.CN129636CN00C5.  
SMBios Version:  3.0
Version: 1.9.0  
Serial number: HTRFRF2 
UUID: 44454C4C-54001052-8046C8C0-4F524632
SKU Number: 0771

Once again thanks for your time and patience.

Mattiwatti, 2020-09-07:
EfiGuard does not create any entries in the BCD store like you may be used to from e.g. UPGDSED. This is because it does not need to do this in order to work.

1. No, that is all. You can leave virtualization enabled (however, you can **not** have HVCI enabled - see the README).
2. EfiGuard will print the following line if PatchGuard was disabled successfully:
`[PatchNtoskrnl] Successfully disabled PatchGuard.`
In general, you can assume that if EfiGuard did not pause the boot process with an error message prompt, PatchGuard is disabled.

You need to boot the system with EfiGuard every time in order to disable PatchGuard. EfiGuard purposely does not persist (i.e. install itself) - you need to activate it for every boot.