Mattiwatti/EfiGuard #23: [Question] int 20h & KiSwInterrupt

AzAgarampur, 2021-03-06:
Hey there, I wanted to know if you have any more information about the handler for `int 20h` that you patch? Your code comments (and other sources) mention that it triggers PatchGuard to "run a verification check." Naturally, I wanted to see how it works by simply making a function that contains
```
Test20 PROC
    int 20h
    ret
Test20 ENDP
```
I didn't expect this to "succeed" - it didn't. It seems like `nt!KiSwInterruptDispatch+0x27` and the instruction `test dword ptr [rdi+994h],100000h` is obviously the culprit, as `rdi` was `0`. I want to know if you know anything more about this interrupt, such as parameters and stuff? I'm assuming that you'll need a PatchGuard context before issuing the interrupt, but I've heard that some anti-cheat for example, like Vanguard, use this interrupt. Do you have any ideas about how they do it?

Mattiwatti, 2021-03-06:
I can't quite remember how I tested this, but as I recall, I allocated a big chunk of pool memory to represent a PatchGuard context, zeroed it and then set the flag at `+0x994` to `0x1000000`. Then I simply patched `KiSwInterruptDispatch` to make `rdi` contain my PatchGuard context instead of the actual kernel PatchGuard one.

Clearly this is not how Vanguard does it, as patching a kernel function to test PatchGuard is a pretty good (and ironic) way to set off PatchGuard. I never investigated further than the above after I found that simply removing the call to `KiSwInterruptDispatch` from the interrupt handler sufficed. Perhaps a Riot employee would care to elaborate? :smile:

NB: I inspected registers on return from the interrupt and there was nothing of note to find. So that this is a "verification call" is only my speculation and may be wrong. It could also simply be a way to check if PatchGuard is initialized, or an easter egg, or... I don't know.

AzAgarampur, 2021-03-06:
Hmm this sounds interesting. I'll see what happens when I try it out. However, before I issue the `int 20h` I'll do something like modify `PsLoadedModuleList` and hide a driver or something and then issue `int 20h` and see if it bugchecks with code `0x109`. I think that'd be a pretty good sign that this is actually a verification routine or if it just does something else.

closed 2023-03-16 by Mattiwatti