Mattiwatti/EfiGuard #24: BSOD on Windows 8.1x64 (6.3.9600.17415 IR5)

assorted, 2021-03-17:
[image]
[image]

Mattiwatti, 2021-03-17:
Hi,

Can you try the latest release (v1.1.1) to see if it fixes the problem? I've had a look at this kernel revision and `KiVerifyScopesExecute` has a prologue that will throw off the old function backtracking algorithm and cause it to overshoot (backwards). I believe this should be fixed with f8ca8c0 though.

assorted, 2021-03-17:
Thank you, I will.

assorted, 2021-03-17:
Quite right, it was an outdated version.

closed 2021-03-17 by assorted