Mattiwatti/EfiGuard #25: Stuck on Windows boot logo

Gold-Phoenix, 2021-04-27:
Just some background info, the flashboot bootmgfw.efi is supposed to be able to boot non CSM, but the GPU drivers never work when CSM is disabled, it boots to desktop but refuses to use the GPU with the error code 21 in device manager.

I tried using EFIGuard on the normal bootmgfw for windows 7 and it gets stuck on the windows boot logo animation thing. I know it loads into the OS because i can hear the windows 7 startup sound.

I would just accept this as defeat, except when I used the flashboot bootmgfw.efi with EFIGuard it boots to the desktop, and loads the gpu (I can tell because aero starts working) except the resolution is tiny and the picture is all messed up.

I'll include a zip containing all the bootmgfw's, a link explaining how the flashboot bootmgfw works, and a link to my motherboard, thanks for your help, if this can actually get solved then I will have a fully working non csm windows 7, which i will be eternally grateful for, thank you.

https://www.prime-expert.com/articles/a21/windows-7-uefi-install-without-csm/

https://www.msi.com/Motherboard/MEG-X570-ACE

[bootmgfw.zip](https://github.com/Mattiwatti/EfiGuard/files/6384146/bootmgfw.zip)

Mattiwatti, 2021-05-01:
Hi,

I had some time to investigate this today, and I ended up with a result of two out of the three machines I tested working with this program.

Results:
- A VirtualBox Windows 7 VM: the flashboot replacement boot manager produced some _extremely_ weird graphical issues during boot, such as the normal-sized VM window suddenly spanning the width of both of my monitors combined. And I could not see the boot selection menu(!), because the display was black during the entire boot process. (The boot menu did work, I just couldn't see it.) But apart from the glitches during the boot process it worked fine in Windows itself.
- A bare metal Coffee Lake Xeon machine with an Asus C246 motherboard: flashboot worked pretty much flawlessly here. I was able to remove the `novesa` and `quietboot` BCD settings and re-enable the `vga.sys` and `vgapnp.sys` drivers  (`vgapnp.sys` was never loaded, but that's normal if you have a 'real' VGA driver installed.), and there were no issues **as far as Windows 7 was concerned**. Booting the Windows 10 installation on my other drive resulted in a black screen of death from the boot manager (maybe because flashboot uses a modified Windows 7 boot manager? I have to say I forgot what the precise error message was, so I don't really know.)

I should note that the above two machines were already previously configured and able to boot Windows 7 without CSM. So for me this was more about verifying that EfiGuard isn't somehow interfering with flashboot. I think it's safe to say that it doesn't. Both machines worked fine, with and without EfiGuard loaded.

This brings me to the last machine I tried - my own one:
- Ryzen 5950X/MSI MPG X570: this gave me exactly the same result you got the first time around. The machine was clearly booted into Windows after a while, but it was still only displaying a black screen. For me though, it didn't make a difference whether or not EfiGuard was loaded.

Final thoughts:
- This is not a bug in EfiGuard (as proven by the first two machines working fine, and the third machine working equally poorly with or without EfiGuard), nor is it a feature I could realistically add, given that I don't even know what the cause is.
- My spider senses tell me it's not a coincidence that both of our motherboards are MSI X570s. This doesn't _necessarily_ mean that this is MSI's, AMD's or AMI's fault, but it's still worth noting. It could also be the case that these platforms differ in a subtle but documented way from some assumptions flashboot is making, in which case it would be a flashboot bug.

My advice would be to take this up with the flashboot authors, as they are probably in a better position to judge who or what is at fault here, since they wrote the code.

Since this is not a bug or something that I can improve in EfiGuard, I will close the issue now. But feel free to reply to it regardless if you get a response to this. As a fellow X570 owner I'd be interested to hear it.

closed 2021-05-01 by Mattiwatti

Gold-Phoenix, 2021-05-01:
Thank you for replying :) people like you give me hope haha

I did try to contact the flashboot authors, no reply yet haha
I have yet to try EFIGuard or Flashboot on a pc that doesnt have an MSI motherboard, i will let you know how it goes once I can go over peoples houses.

As for your MSI motherboard were you able to get EFIGuard to boot past the windows boot logo? or is it just an MSI thing to get stuck there?

Best regards :)

Mattiwatti, 2021-05-02:
To clarify, if it is indeed a problem with the platform firmware and not with flashboot, it's not certain that it's *MSI* that's the problem. If this also happens on other X570 motherboards, I would say it's more likely that it's an issue with AMI's (American Megatrends) BIOS framework for the platform, or (less likely) AMD's AGESA.

Most motherboard manufacturers write very little or none of code in the BIOSes they release, apart from some renaming of features and branding (logo's, UI design and such). Instead they source a 'platform base' framework for a BIOS from another vendor. Major companies making these are AMI, Phoenix and Insyde. This means that there are actually relatively few major differences between e.g. a Gigabyte and an MSI BIOS for the same platform.

> As for your MSI motherboard were you able to get EFIGuard to boot past the windows boot logo? or is it just an MSI thing to get stuck there?

I have the boot logo disabled, so I can't say for sure whether I got stuck booting 'at the logo' or if I indeed booted into Windows. But my PC is quite noisy, and it went quiet after some time (about the same time as you'd expect a normal boot to take). So I'm pretty sure Windows booted successfully, but that I just couldn't see it because there were no graphics.

Gold-Phoenix, 2021-05-02:
Thank you for insight into how BIOS's are made, I never knew any of that.

And as for not knowing if it booted into the OS, if you have the login sound enabled just plug a speaker into the jack and you'll hear it, and that's a shame. I'll post here again if I get any different results from my friends computers, they're all x570 boards.

Thank you again :)