Mattiwatti/EfiGuard #26: Trying to add to Boot Menu

atomik00, 2021-04-27:
Im not sure how to go about this but i want to add the EFIGuard option to the boot menu without having to go through efishell. is this even possible? i added an entry to bcdedit that shows up and is pointing to the loader but it blue screens with the recovery page everytime. if i put on a usb stick and boot the usb it works fine so im not sure what else to try so i can add this to the boot options without having to go into efishell.

Mattiwatti, 2021-04-30:
The answer to this question depends on what exactly you mean by
> i want to add the EFIGuard option to the boot menu

The reason I say this is because the answer is yes, you can in fact add EfiGuard as a boot entry using `bcdedit`. But this probably is not what you actually want. First off, here's how to do it:
```
bcdedit /copy {bootmgr} /d "EfiGuard"
```
(bcdedit will now print a GUID for the new entry, which we'll call `<guid>`)
```
bcdedit /set <guid> path \EFI\Boot\Loader.efi
bcdedit /set {fwbootmgr} displayorder <guid> /addfirst
```
If you reboot now, EfiGuard will boot and will work perfectly fine, booting Windows successfully.

But this is probably not what you had in mind when you said "the boot menu", because running the above commands is essentially the same thing as adding EfiGuard as a boot option via the UEFI shell. In other words, it will be added to the _firmware_ boot options, not the Windows boot manager boot options. (But maybe that **is** what you meant, I don't know.)

It is not possible to have the Windows boot manager boot EfiGuard, due to:
1. The way EfiGuard works - it must be the first in a chain of any Microsoft images that are loaded, because each stage of patching the next image is dependent on EfiGuard having patched the boot manager.
2. The highly restrictive policies that MS has in place for what the Windows boot manager will actually boot. In theory, I could have made EfiGuard a replacement for `winload.efi` instead of `bootmgfw.efi`, and have the ability to add it to the Windows boot menu. But in practice this is not realistic because of things like secure boot. But even without secure boot, and with test signing and boot debugging enabled, the Windows boot manager will refuse to boot images whose certificate it cannot verify.

Here is what a quick attempt to load EfiGuard from the Windows boot menu looks like:

[image]
This is with test signing and boot debugging enabled, secure boot disabled, and the file being signed (albeit self-signed).

(Amusingly enough, this error would not have occurred if EfiGuard had already been loaded, because it patches the function that is responsible for this verification. But I can hardly make EfiGuard a dependency of itself.)

So what should you do instead? Instead of using the Windows boot manager, install a boot manager that doesn't suck, and then use that to boot whatever OS you want, including Windows.

There are quite a few freely available UEFI boot managers out there, but I'm not very familiar with any of these so I don't want to give a recommendation there. But I can tell you that GRUB works perfectly fine with EfiGuard. GRUB can boot pretty much any EFI image, which means you can use it to chainload the Windows boot manager to boot Windows, or start EfiGuard, which will then boot Windows. For what it's worth, the reverse works too: if you accidentally boot EfiGuard on a system that has Linux installed but not Windows, EfiGuard will start GRUB instead (and unload itself) and GRUB will work fine.

See #21 for a basic `grub.cfg` entry that will boot EfiGuard.

atomik00, 2021-05-04:
Thank you for the response. I was referring to Windows Boot Manager, it was already showing in the boot options from my usb but not from a regular hard drive.  I was not aware of the command above changing the name though thats nice. Thanks for that.

closed 2023-03-16 by Mattiwatti

ibay770, 2024-06-26:
> The answer to this question depends on what exactly you mean by
>
>
> The reason I say this is because the answer is yes, you can in fact add EfiGuard as a boot entry using `bcdedit`. But this probably [...]

Thank you for your work and explanation! Alternatively, is there a way to patch those files offline (like from an iso) so there is no need to do this every time?

ibay770, 2024-10-10:
Is there  a script that can do this automatically?