Mattiwatti/EfiGuard #27: Compiling Questions

atomik00, 2021-05-04:
I am looking into using this patch to apply my own driver on boot. is this possible to load another driver at boot or would i have to work within the full standard EFIGuard source you have in order to make something like that work? 
I only ask because i have tried so many times to setup EDK and i cannot for the life of me get it to work let alone looking at source code lol. 
I am very knowledge in drivers as i currently have one now that works fine when loaded through windows im just trying to find a way to patch windows and run this driver on boot rather then loaded once in windows. if that makes sense.

atomik00, 2021-05-08:
i already load a driver on boot currently using some edk libs i built within linux so its easy to compile without issues. i was looking at your project as a way around patch guard and disabling dse so that i can run another driver within windows without issues. the problem is it would be a 2 step process currently which would be loading my driver then loading the patched os what i was trying to see is if i could chain load them somehow so in grub i would just select "EFIGUARD" and that loads both. does that make sense?

Mattiwatti, 2021-05-14:
Yes, Grub can start EfiGuard and vice versa. (Although if EfiGuard detects a non-Windows OS being booted, it will unload itself.) How and when your own UEFI driver is loaded is up to you.

See #21 for a basic `grub.cfg` entry that will boot EfiGuard.

closed 2023-03-16 by Mattiwatti