Mattiwatti/EfiGuard #29: Startup

IvankoEanko, 2021-05-24:
Is it possible to do something so that I do not have to turn this entire operation in the boot every time?

Mattiwatti, 2021-05-24:
Yes, you can install EfiGuard on the Windows ESP partition and make a boot entry for it. See [this comment on issue #2](https://github.com/Mattiwatti/EfiGuard/issues/2#issuecomment-478998015) for instructions.

IvankoEanko, 2021-05-24:
I have only network devices in UEFI.

IvankoEanko, 2021-05-24:
What is X?
[image]

IvankoEanko, 2021-05-24:
I am stuck on this, I try to do it 100 times but it doesn't work. I don't understand what X is, I thought it was a flash drive. In my case, this is E, but if I type E it says that the Folder is not empty. Also, the cd command does not work in effi for some reason. Is there a video about it anywhere?

IvankoEanko, 2021-05-24:
I have done it. Ty

closed 2021-05-24 by Mattiwatti