Mattiwatti/EfiGuard #30: Startup problem 2

IvankoEanko, 2021-05-24:
I did everything according to the guide, a message was displayed for me, I even checked the dump to see if my EfiGuard remained there in 1st place. But when you reboot, it disappears from startup

Mattiwatti, 2021-05-24:
I'm assuming (based on the previous issue you created) that you're referring to installing EfiGuard on the Windows ESP...

If `bcfg boot dump` in the UEFI shell shows EfiGuard, then it exists in the BIOS boot entries. Check your BIOS startup options. Some firmwares get confused when there are multiple boot entries for the same drive, and/or will add the new entry but place it at the back. Look for 'boot entry order', 'BBS disk order', 'disk boot order' or similar. Chances are you will find at least two entries:
- Windows Boot Manager
- EfiGuard

If so, swap their order so that EfiGuard is first, but do not remove the Windows Boot Manager entry because EfiGuard needs it to find Windows.

Mattiwatti, 2021-05-25:
I'm guessing that by 'EfiGuard dump', you mean the output of `bcfg boot dump` in the UEFI shell...

> I put it in 0 place, after Windows

Either you put it in 0th place (i.e. it's the first entry), or you put it after Windows, not both. EfiGuard should be listed *before* Windows in the output of `bcfg boot dump`.

Again, check your BIOS options. What `bcfg` tells you does not always correspond to what the BIOS will actually do, especially if there are multiple boot entries on the same drive.

IvankoEanko, 2021-05-25:
[image]
[image]

IvankoEanko, 2021-05-25:
But it disappears after reloading one more time

IvankoEanko, 2021-05-26:
> I'm guessing that by 'EfiGuard dump', you mean the output of `bcfg boot dump` in the UEFI shell...
>
>
> Either you put it in 0th place (i.e. it's the first entry), or you put it after Windows, not [...]

i dont know. I'm downloading this from a USB stick and I noticed that EfiGuard has optional N and all others have Y. Maybe this is the case?

Mattiwatti, 2021-05-26:
> i dont know. I'm downloading this from a USB stick and I noticed that EfiGuard has optional N and all others have Y. Maybe this is the case?

No, this is unrelated and a side effect of the entry being added by `bcfg`.

Your screenshot is showing a path to EfiGuard that is on a USB drive, not on the Windows EFI partition you are (as I understand it) trying to use... you either have EfiGuard installed in the wrong place, or you are invoking `bcfg` from the wrong drive or with the wrong path.

Furthermore, your USB drive is formatted as MBR and not as GPT. (This is OK for USB drives on some motherboards, but there are also plenty of motherboards that will not accept this.)

Both of the above _could_ be reasons why your BIOS is deciding to throw out the boot entry. I really don't know.

IvankoEanko, 2021-05-26:
> No, this is unrelated and a side effect of the entry being added by `bcfg`.
>
> Your screenshot is showing a path to EfiGuard that is on a USB drive, not on the Windows EFI partition you are (as I [...]

That is, I need to shove your files to the Windows boot?

Mattiwatti, 2021-05-26:
If you want to add EfiGuard as a permanent boot entry, then yes, I would recommend doing so. It may be possible to add permanent USB boot entries by filename on some motherboards, but in general I would say it is not reliable.

In the [previous issue you made](https://github.com/Mattiwatti/EfiGuard/issues/29), you quoted the exact steps needed to do this. I don't know where the confusion is coming from.

IvankoEanko, 2021-05-26:
> If you want to add EfiGuard as a permanent boot entry, then yes, I would recommend doing so. It may be possible to add permanent USB boot entries by filename on some motherboards, but in general I [...]

I do not have access to Windows boot, in the last topic I thought that everything was fine, because at startup I got an inscription, after another restart, the inscription disappeared.

IvankoEanko, 2021-05-28:
> If you want to add EfiGuard as a permanent boot entry, then yes, I would recommend doing so. It may be possible to add permanent USB boot entries by filename on some motherboards, but in general I [...]

When I start it manually, everything works, but if I want to make it as autoload, then Windows is loaded
[image]
[image]
I installed it and it boots fine, but after rebooting the system.
[image]

IvankoEanko, 2021-05-28:
My usb is 
[image]

IvankoEanko, 2021-05-29:
Solved. Enable Legacy mode and disable boot mod.

closed 2021-05-29 by IvankoEanko