Mattiwatti/EfiGuard #34: Pre-boot custom code

NexSqaud, 2021-06-15:
EfiGuard can patch a Windows, but can it run some code before run a Windows, and shutdown PC without "BSOD"? For example, password boot lock. And if yes, can you give a link to some documentation?

Mattiwatti, 2021-06-24:
> EfiGuard can patch a Windows

Yes...

> but can it run some code before run a Windows

That's literally how it works. By executing code before Windows does. I think you need to be more precise about what you want here.

> and shutdown PC without "BSOD"?

Do you want to execute code, shut down your PC, or both? And at what moment(s) should these actions take place? I get that you don't want it to BSOD, but I don't think that's really relevant for the question. If you were more specific (a *lot* more), I would be able to tell you whether EfiGuard can do what you want or not (i.e. 'yes' or 'no'). Whatever the question, I highly doubt that the answer is that EfiGuard can do it, but somehow only with an added BSOD.

> For example, password boot lock

What kind lock? The firmware/BIOS password lock, a drive encryption (Bitlocker/Veracrypt/...) lock, or a Windows user password lock? And what is it that you want EfiGuard to do with this lock (or password)? Lock it? Unlock it? Find the password? In all cases the answer is almost certainly **no**.

> And if yes, can you give a link to some documentation?

There is [README.md](https://github.com/Mattiwatti/EfiGuard/blob/master/README.md). That is the only EfiGuard documentation there is. But it covers:
- What EfiGuard does (or 'can do')
- What EfiGuard can not do
- How to use EfiGuard
- How EifGuard works
- How to compile EfiGuard

In my opinion the `README` is pretty thorough, and should answer most questions people may have about EfiGuard, and if not, point them to the right resources to find out for themselves.

In this case, I *believe*, from your vaguely worded questions regarding various entirely unrelated things, that you are
- Asking whether EfiGuard can do something that it was not at all designed or intended to do, such as executing arbitrary code at some specific moment in the boot process. The answer to this is **almost certainly no**, but unless you clarify what you actually want to do, I don't know.
- Asking whether there is documentation available regarding `X`, where `X` is whatever your desired feature is. The answer to this is **yes**, and what you are asking about can be found in one or more of these sections of `README.md`: 'what EfiGuard does (or can do)', 'what EfiGuard can not do', and 'how EfiGuard works'. That is: if `X` is listed as one of the things EfiGuard can do (unlikely), then the answer to the first question is also actually yes. If `X` is listed as one of the things EfiGuard can not do, then the answer to the first question is still no. If `X` is listed in neither of these two sections, the answer to the first question is probably "no, but maybe it could". To determine whether EfiGuard **could** do `X`, carefully read the 'how EfiGuard works' section to determine the answer for yourself. If the answer is "no", then you are done. If the answer is "yes, EfiGuard could do this", move on to the 'how to compile EfiGuard' section and help yourself.

NexSqaud, 2021-06-24:
> Do you want to execute code, shut down your PC, or both?

Execute code and as result shutdown PC or run Windows.

> What kind lock?

Just a lock on EfiGuard that blocks Windows from starting before correct password doesn't written. A BIOS/Firmware variant not suitable at my case.

Mattiwatti, 2021-07-01:
> Execute code and as result shutdown PC or run Windows.

> A lock on EfiGuard that blocks Windows from starting before correct password doesn't written. A BIOS/Firmware variant not suitable at my case.

After several readings, I am now understanding this to mean: 'I want to execute some piece of code that blocks Windows from loading until the user enters a password. If the password is correct, continue booting Windows. If it is not, shut down the PC instead'. Or, in pseudocode:
```C++
void CheckAllowedToBootWindows()
{
    string Password = GetUserPasswordInput();
    if (!VerifyPassword(Password))
    {
        Shutdown();
    }
}
```
If this interpretation of your question is correct, the answer is **yes**, EfiGuard **could** do this (but it doesn't, because it clearly has nothing to do with what EfiGuard was written to do).

I have already told you what to do in this case in my previous reply, but here is the specific quote:
> Move on to the 'how to compile EfiGuard' section and help yourself.

I'd feel more inclined to help you if your request/proposal wasn't of dubious value at best, and clearly indicative of malicious intent at worst.

> A BIOS/Firmware variant not suitable at my case.

Then why are you asking this? What do you think UEFI is?

Reply not necessary, closing this.

closed 2021-07-01 by Mattiwatti