Mattiwatti/EfiGuard #37: Question

h8suga, 2021-08-05:
Its possible on old BIOS not UEFI?
my system dump:
`SystemBootEnvironmentInformation:
        - BootIdentifier: {13299921-eaa4-11eb-9b90-d99f30e81197}
        - FirmwareType: BIOS
        - BootFlags: 0x1

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000`

Mattiwatti, 2021-08-10:
No. Legacy/BIOS mode support is not supported by EfiGuard, and there are no plans to add support for it. Sorry.

closed 2021-08-10 by Mattiwatti

JuniorJPDJ, 2021-08-10:
It actually is possible using workaround:
You can boot EFI on your legacy BIOS machine.
Use [Clover](https://github.com/CloverHackyColor/CloverBootloader) (preferable and easier to setup) or Tianocore Duet for this purpose.