Mattiwatti/EfiGuard #38: TItanhide.sys  It doesn't work

wtf3824, 2021-09-06:
windows 10 x64 1511 ，at work。 √
windows 10 x64 1607, It doesn't work。   X

Please help me.  

I need version 1511 above running

wtf3824, 2021-09-06:
https://github.com/mrexodia/titanhide

Mattiwatti, 2021-09-07:
'It doesn't work' is not a helpful description of any problem. If TitanHide doesn't work, I would say you should make an issue on the TitanHide repository. But since you made one here instead, I'm assuming you've got an issue with EfiGuard.

Please **describe what the issue is** and how to reproduce it.

sunbeam906, 2022-07-09:
I believe what he's saying is EFIGuard is outdated for the latest Window 10 builds. There's been several cumulative patches since May 2021 (date of latest EFIGuard release) which broke its functionality. I've reported this already and I understand it's most likely in the works, so the user reporting this should have patience ;)

AntonKukoba1, 2022-08-25:
@sunbeam906 any news? I'd like to run https://github.com/everdox/InfinityHook but I get KERNEL SECURITY CHECK failure all the time... It's Windows 10 x64 21H2

sunbeam906, 2022-09-03:
No news. I recommend sticking to a version you know EFIGuard has worked on and prevent Windows Update from running; if you want to use this method still. Otherwise, I won't go into the semantics of why you want to keep your OS up-to-date with Windows Update... Take it or leave it :) Cheers.