Mattiwatti/EfiGuard #39: Can we fake the presence of a TPM using efiguard?

DavidXanatos, 2021-10-05:
Can we fake the presence of a TPM using efiguard?
With this super irrational demand of MSFT to have a TPM in order to install windows 11, I was thinking if we can somehow reliably fake windows into thinking there is a TPM when there is 
1. none or 
2. a cheap emulation that just saves everything to a plain text file on the efi partition, muhahahahah...

e-t-l, 2022-03-04:
Not sure if EfiGuard can do it, but you can easily do it with [https://github.com/AveYo/MediaCreationTool.bat/](https://github.com/AveYo/MediaCreationTool.bat/)

Mattiwatti, 2022-03-29:
> Can we fake the presence of a TPM using efiguard?

Short answer: no.

Slightly longer explanation: see [here](https://github.com/Mattiwatti/EfiGuard/issues/45#issuecomment-992899252).

closed 2022-03-29 by Mattiwatti