Mattiwatti/EfiGuard #40: Can we fake usage of Secure Boot using efiguard?

DavidXanatos, 2021-10-05:
Since MSFT demands Secure Boot for windows 11 to be on in order to install, and this is a irrational demand.
I was wondering if we can sue a efiguard to fake the secure boot starts variable such that on a system without secure boot windows wil think it was booted securely?

Cheers
David X.

smokeSH, 2022-03-13:
Just a side note: Windows 11 doesn't require Secured Boot to be turned on to install, it only requires Secure Boot to be available in general.

DavidXanatos, 2022-03-14:
I am aware of the current state of things, but its always good to be one step ahead of your enemy.

Mattiwatti, 2023-03-16:
Sorry for the delay in response.

Yes, this is possible, but generally speaking, emulating Secure Boot is not a goal of EfiGuard (unless there is a case for this I didn't think of and which you think would benefit the project of course - if so, feel free to reopen this with a comment and I'll reconsider).

See image (I've highlighted areas of interest):
[image]
This was done by simply reusing the existing `SetVariable` hook code in `EfiGuardDxe.c` and repurposing it for a `GetVariable` hook on the Secure Boot variable.

The main issues with this approach are:
- Increased complexity and chance of bugs due to the added runtime variable hook.
- EFI variable measurements (performed before EfiGuard is loaded) will be recorded by the firmware into the TPM event log, making it not very useful to do this in practice. This can be seen in the screenshot above, where due to this, for example (among other things) BitLocker will not be usable with the default configuration settings due to PCR7 not being usable.

Code for reference:
```
#include <Guid/ImageAuthentication.h>

// ...

EFI_STATUS
EFIAPI
HookedGetVariable(
    IN CHAR16 *VariableName,
    IN EFI_GUID *VendorGuid,
    OUT UINT32 *Attributes OPTIONAL,
    IN OUT UINTN *DataSize,
    OUT VOID *Data OPTIONAL
    )
{
    if (VariableName != NULL && VariableName[0] != CHAR_NULL && VendorGuid != NULL && DataSize != NULL &&
        CompareGuid(VendorGuid, &gEfiGlobalVariableGuid) &&
        StrnCmp(VariableName, EFI_SECURE_BOOT_MODE_NAME, (sizeof(EFI_SECURE_BOOT_MODE_NAME) / sizeof(CHAR16)) - 1) == 0)
    {
        if (Attributes != NULL)
        {
            *Attributes = (EFI_VARIABLE_BOOTSERVICE_ACCESS | EFI_VARIABLE_RUNTIME_ACCESS);
        }
        CONST UINTN InDataSize = *DataSize;
        *DataSize = sizeof(BOOLEAN);
        if (Data != NULL && InDataSize >= sizeof(BOOLEAN))
        {
            *(BOOLEAN*)Data = SECURE_BOOT_MODE_ENABLE;
        }

        if (InDataSize < sizeof(BOOLEAN))
            return EFI_BUFFER_TOO_SMALL;
        return Data != NULL ? EFI_SUCCESS : EFI_INVALID_PARAMETER;
    }

    return mOriginalGetVariable(VariableName, VendorGuid, Attributes, DataSize, Data);
}
```

closed 2023-03-16 by Mattiwatti

anon867, 2025-08-07:
> Sorry for the delay in response.
>
> Yes, this is possible, but generally speaking, emulating Secure Boot is not a goal of EfiGuard (unless there is a case for this I didn't think of and which you [...]

Not sure if this code implementation is still relevant now, but I did try to add this and it BSOD'd me on 22H2. Was wondering if you had any insight?