Mattiwatti/EfiGuard #43: Using EFiGuard with Rainbow, computer only restarts can't shut down.

0pLuS0, 2021-11-02:
Hello,

First I want to say thank you for your development of Efiguard, it really helps solving one of my problems when using Rainbow, which is constant BSOD, but now with EfiGuard Windows won't shut down. Every time I try to shut the computer down it just keeps restarting.

Luckily I have a dual boot box also running Linux, so for now, I boot into Linux and shut the computer down from there.

I hope there's a solution for this?

These are my Specs.

Windows 10 2004

Motherboard
ASUS Maximus VIII Hero

PSU
EVGA SuperNOVA 550 G3

RAM
Corsair Vengeance LPX 16GB (2x8GB) DDR4 DRAM 2133MHz C13

CPU
i7 -6700K

Video
EVGA GeForce GTX 1060 06G-P4-6262-KR

Monitor
MSI Optix G24C - 144hz

THANKS

0pLuS0, 2021-11-03:
After posting this, I made these changes to my system and I was able to shut down the computer, but then I had a BSOD, so now I need to figure out if it's possible to stop getting BSOD.

1. I disabled Fast Start
2. For the network adapter I disabled; Allow this device to wake the computer
3. In the registry, 'HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows NT\CurrentVersion\Winlogon' 
PowerDownAfterShutdown” entry in the right pane I changed from 0 to 1

Mattiwatti, 2021-11-07:
Hi,

Thanks for the detailed report, and thanks for following up with the additional info which will (hopefully) help me narrow it down. I have only one question: what is "Rainbow"? 😄 That would be relevant information for sure, since you mention that you use EfiGuard because of this. Do the two problems (PC does not shut down / Windows BSODs) occur even when you use only EfiGuard and not both?

I am aware of a *possible* power related issue when resuming from ACPI S4 (see [this comment on #12](https://github.com/Mattiwatti/EfiGuard/issues/12#issuecomment-625496546)  for more information) - but this was only reported once and later reported fixed on the user's machine. What you are describing sounds unrelated to this issue to me. I would sooner expect that something or someone is preventing you from shutting down via some other method such as hooking the EFI system reset service, or writing data to the `OSIndications` UEFI variable to achieve a similar effect. (I don't think `OSIndications` can actually prevent a shutdown, but I may be wrong there. It can definitely be used for things that are similar to whare you are describing.)

I have a machine with hardware roughly similar to yours and tried to repdroduce the issue on that by basically doing the steps in your second post in reverse, but could not reproduce it, even with 3 NICs all having WOL enabled. I should add that the power setting screen (at least on my PC) stated pretty clearly that rebooting is not covered by 'fast boot':

[image]

0pLuS0, 2021-11-09:
I was told, this has something to do with signed code and unsigned code, causing these issues. I believe Rainbow is unsigned and maybe EFIGuard are too.

This is the Github Repo for Rainbow.

https://github.com/SamuelTulach/rainbow

I noticed that if I close down apps in a certain order I seem to be reducing the risk of getting a BSOD.

Mattiwatti, 2023-03-16:
Yes, EfiGuard is not signed if you used a precompiled release from GitHub.

I still don't see the relevance of this Rainbow project here nor do I get what you are trying to do. Closing this unless you can provide some information (like a crash dump...) that implicates EfiGuard in these BSODs and not a completely unrelated project.

closed 2023-03-16 by Mattiwatti