Mattiwatti/EfiGuard #45: Windows 11

NightRaidGamer, 2021-12-13:
hello i would like to know if this bootkit is compatible with windows 11?

Mattiwatti, 2021-12-13:
Hi,

Yes, EfiGuard works fine with Windows 11 (at the time of writing).

I should probably update the README to reflect this, but I have also received some questions in the past that, to me, seem to equate "Windows 11 support" with some secret new tech that can emulate a TPM (in particular in a way that would survive attempts at remote attestation).

So to clarify:
- Every feature of EfiGuard works on Windows 11, the same way they did on Vista SP2 through 10.
- No changes have been made or will be made to bypass the TPM requirements of Windows 11.

closed 2021-12-14 by NightRaidGamer

e-t-l, 2022-03-04:
> No changes have been made or will be made to bypass the TPM requirements of Windows 11.

@Mattiwatti I mentioned this in my answer https://github.com/Mattiwatti/EfiGuard/issues/39#issuecomment-1059426371 (and that issue could probably be closed); you may want to direct people to https://github.com/AveYo/MediaCreationTool.bat/ as a solution to TPM circumvention.

Mattiwatti, 2022-03-29:
Thanks for the suggestion, but I don't really feel comfortable recommending some "universal media creation tool" in the project's `README` just because Windows 11 has introduced additional anti-consumer measures and people (including myself) would like to get rid of them.

EfiGuard is primarily intended for advanced users who, well, know what they are doing by taking fairly drastic measures (namely disabling PatchGuard and DSE). If Windows 11's TPM requirements are an obstacle to using EfiGuard, perhaps reconsider whether you need EfiGuard to begin with, and if there isn't some more suitable replacement for whatever it is you are trying to do.