Mattiwatti/EfiGuard #49: error 4000: Instance of library class [VariablePolicyHelperLib] is not found

Lamed-D, 2022-03-26:
C:\edk2>build -a X64 -t VS2017 -p EfiGuardPkg/EfiGuardPkg.dsc -b RELEASE
Build environment: Windows-10-10.0.22000-SP0
Build start time: 22:50:23, Mar.26 2022

WORKSPACE        = c:\edk2
EDK_TOOLS_PATH   = c:\edk2\basetools
EDK_TOOLS_BIN    = c:\edk2\basetools\bin\win32
CONF_PATH        = c:\edk2\conf

Processing meta-data .PYTHON_COMMAND   = py -3

Architecture(s)  = X64
Build target     = RELEASE
Toolchain        = VS2017

Active Platform          = c:\edk2\EfiGuardPkg\EfiGuardPkg.dsc

build.py...
c:\edk2\EfiGuardPkg\EfiGuardPkg.dsc(...): error 4000: Instance of library class [VariablePolicyHelperLib] is not found
        in [c:\edk2\MdeModulePkg\Library\UefiBootManagerLib\UefiBootManagerLib.inf] [X64]
        consumed by module [c:\edk2\EfiGuardPkg\Application\Loader\Loader.inf]

- Failed -
Build end time: 22:50:23, Mar.26 2022
Build total time: 00:00:00

Lamed-D, 2022-03-27:
Closed 
somehow solved

Lamed-D, 2022-03-27:
0

closed 2022-03-27 by Lamed-D

Mattiwatti, 2022-03-29:
Thanks. This was actually an issue introduced by a recent change in EDK2.

I've fixed the build for the current EDK2 master branch (and anything >= `edk2-stable202202`). I'm guessing you probably resolved the issue by building with an older version of EDK2, which also works.

anl135, 2022-07-28:
Hi Lamed,
Could you tips how you solve this problem? Thank you!

Lamed-D, 2022-08-02:
> Hi Lamed, Could you tips how you solve this problem? Thank you!

I download older versions