Mattiwatti/EfiGuard #50: load EfiGuardDxe.efi   After the black screen

CrowMouth, 2022-04-02:
https://user-images.githubusercontent.com/99059308/161390083-f486d975-96d1-4492-b10c-648235b37599.mp4

I am foolish and ignorant, and I cannot find the cause of this problem.

In UEFI SHELL, after load efiguarddxe. efi, it shows a black screen soon after loading successfully (no task text), passing by the kind big guy,
[image]
 can you leave your valuable experience, but I can only express my gratitude (this is ashamed), Amon!

Mattiwatti, 2022-04-02:
Hi,

Google Translate, or whatever it is you're using, doesn't seem to be working very well, so it's a bit hard to decipher what the issue is. I'm guessing that after `load EfiGuardDxe.efi`, the screen stays black and the computer hangs like that indefinitely?

If so, you can try
`load -nc EfiGuardDxe.efi`
instead to see if that helps.

CrowMouth, 2022-04-03:
You are right, I am using the translation, the poor translation should be my expression of the problem, let you laugh.

I didn't expect you to reply so quickly, and your understanding is accurate, of course, only about 30 minutes indefinitely.

load -nc EfiGuardDxe.efi solved this problem and I was very excited, although I didn't know what it meant.

If you can, I would like to please do not delete this problem, let more weaklings like me see, help them.

Thank you for your guidance

Mattiwatti, 2022-04-03:
Sure, no problem. I am going to close this though as it sounds like the issue has been resolved.

closed 2022-04-03 by Mattiwatti