Mattiwatti/EfiGuard #51: EfiDSEFix native support

eebssk1, 2022-04-14:
Care to add a project configuration to EfiDSEFix that as a SUBSYSTEM:NATIVE application so it can be used in BootExecute?
The implementation seems already ready for this, just need to change a little thing to work.

Mattiwatti, 2022-04-19:
Hi,

Interesting idea, I hadn't considered this use case before.

I can definitely do this for the EfiDSEFix project if you want (I guess I could call the configuration 'Release (subsystem native)'), but in essence all this would be changing would be, well, the subsystem type. I'm a little more hesitant to make the same change on the solution level since it would (appear to) affect all of the projects in the solution, even though the only change would be to the EfiDSEFix project and not the EFI modules. Given that you are asking this question I assume you already know how to do the former and that this is more about not needing to maintain a private fork for such a minor change (which is understandable). Would it still be helpful if I made the change only for the EfiDSEFix project but not the solution file? Or would it need to be both?

NB: I haven't tested this myself so I'm not 100% sure, but from what I remember there would likely need to be additional changes in order for EfiDSEFix to work as BootExecute image. The only one that comes to mind would be any `Printf` calls (= `WriteConsoleW` from kernel32.dll) - these would probably have to be removed (easiest) or be changed to `NtDisplayString` calls.

eebssk1, 2022-04-19:
I think only make the change in the EfiDSEFix project is enough with a note in the readme.

It's tested that remove link to kernel32.lib and the printf function is enough to make it work.

Mattiwatti, 2022-04-21:
Done.

I tested this using `NtDisplayString` and it works fine, so I've made EfiDSEFix use that as a replacement printf.

commit d128e2a3 (2022-04-21): EfiDSEFix: add /SUBSYSTEM:NATIVE project configuration

Closes #51
closed 2022-04-21 by Mattiwatti