Mattiwatti/EfiGuard #52: BSOD on EfiDSEFix.exe -d

hassenhamdi, 2022-04-20:


Mattiwatti, 2022-04-20:
I'm going to need some more info than this.

Mattiwatti, 2022-04-20:
Thanks, much better. Unfortunately I can't reproduce this either on Windows 10.0.19041.1645 or on Windows 11 insider.

You made a screenshot of your recently installed updates - am I assuming correctly that this BSOD didn't occur prior to installing those yesterday? I'd install them myself to check, but they aren't listed on the Windows Update catalog. In fact all I can find about these KBs on Google is threads about people having issues with them... 2 years ago...

If you could post the kernel dump file that'd also be helpful, since currently I can't see what type of fault (read/write/execute) is occurring at the point where the page fault happens.

hassenhamdi, 2022-04-20:
it is the first time using this tools

hassenhamdi, 2022-04-20:
I downoload  it just today

hassenhamdi, 2022-04-21:
[042122-16468-01 - Copie.zip](https://github.com/Mattiwatti/EfiGuard/files/8534920/042122-16468-01.-.Copie.zip)
this is the minidump file I could not find a larger dump

hassenhamdi, 2022-04-24:
did you find a solution for the issues ?

Mattiwatti, 2022-04-26:
> did you find a solution for the issues ?

No.

Your crash dump is **extremely** odd, because it indicates that this is not a page fault as I was expecting it to be (or otherwise perhaps a general protection fault), but rather a 'segment not present fault' (these are named '#NP' in the Intel manual). This is not something that will ever occur in long mode (x64 mode, which is what you were in at the time of the crash) in any normal configuration.

My best guess is that you have some sort of hypervisor running and that it is either buggy, or else injecting the #NP on purpose. This could be something 'benign' such as an antivirus (insofar as AVs can be called benign) or something more sinister/malicious. Again though, this is only a guess. I don't really know what is causing this, but it's the only plausible explanation I can come up with.

Mattiwatti, 2023-03-16:
Any chance you could retry with 2f4a666? See attached if you can't or don't want to compile the driver yourself, since there is no new release with this change yet: [EfiGuardDxe.zip](https://github.com/Mattiwatti/EfiGuard/files/10995223/EfiGuardDxe.zip)

Note that what this commit fixes does not really have anything in common with your issue as far as I can tell, other than being a BSOD while a hypervisor is present, so this is a bit of a stab in the dark.

Mattiwatti, 2023-10-04:
Closing this absent any response and with no way to reproduce, feel free to reopen if you figure out how to reproduce this more reliably.

closed 2023-10-04 by Mattiwatti