Mattiwatti/EfiGuard #56: EFIGuard Issue Booting Into Windows

k1osk, 2022-06-25:
Hi,

I have an issue when booting from a FAT32 USB - Loader.config.efi deleted, Loader.efi renamed to "bootx64.efi", after "Successfully patched ntoskrnl.exe." has been displayed on BIOS, I do not boot into windows as expected.

Instead my monitor does not receive signal (sleeps) and my computer remains on.
I have made sure BIOS mode displays UEFI before boot. Gigabyte motherboard and my windows version is: 10.0.19044 Build 19044

Any help would be greatly appreciated! Thanks for your work.

Mattiwatti, 2023-02-27:
Sorry, I have absolutely no idea what this could be. Did you manage to resolve the problem?

My only guess would be a firmware issue (meaning your motherboard BIOS). If so, it could potentially be resolved by updating to the latest BIOS version. But I'm not sure I really buy this explanation to begin with since any hardware or firmware related issue would normally have been obvious long before the moment the kernel is being loaded by the bootloader.

leele2, 2023-05-07:
Having the same issue also with a gagabyte board

MrDebugger0, 2024-03-30:
Me too, also gigabyte motherboard. It just constantly keeps loading EfiGuard over and over. and doesn't boot the windows.

acoul, 2024-06-13:
FWIW, in order to get a useful answer you have to provide as detailed as possible system setup & information

additionally: 1 image (about the problem) = 1000 words