Mattiwatti/EfiGuard #58: system thread exception not handled error on windows 11

UtkuYLMZ, 2022-07-02:
I'm trying to boot from a USB stick, but I get a BSOD with an error "system thread exception not handled".

Mattiwatti, 2023-02-27:
Sorry for the late response. If I had to guess I would say I probably skipped this issue because I can't really do much with this info as is.

*When* do you get this BSOD? At boot? When running `EfiDSEFix.exe -d`? I'll just assume the latter for now.

Please upload a crash dump file so I can take a look at it. (A 'kernel' or 'automatic' minidump will probably suffice.) Also, please post the full output of `EfiDSEFix.exe -i`.

Mattiwatti, 2023-03-16:
I'm *assuming* this is the same BSOD as #57, which has been fixed.

If that is incorrect and the BSOD still occurs with 2f4a666, feel free to reopen.

closed 2023-03-16 by Mattiwatti