Mattiwatti/EfiGuard #59: Win11 21H2 BSOD on ''EfiDSEFix.exe -d'' Command

spIoopy, 2022-07-06:
## BlueScreenView

[image]

## Event Viewer

[image]

`The computer has rebooted from a bugcheck.  The bugcheck was: 0x0000003b (0x00000000c0000005, 0xfffff8038062fd80, 0xfffff50d3b6f3cc0, 0x0000000000000000). A dump was saved in: C:\WINDOWS\MEMORY.DMP. Report Id: 49033993-ef6d-4d11-ba1a-d2dbbf2c7a86.`

[image]

## BSOD

SYSTEM_SERVICE_EXCEPTION

0x00000000C0000005
0xFFFFF80264FAFD80
0xFFFFFD831774ECC0
0x0000000000000000

## Extra Info

First time using, tried the command 5 times, still the same thing!

Mattiwatti, 2022-08-12:
Sorry for the delay. Since this is Windows 11: do you have Hyper-V enabled? If yes, you should disable it, see #57. If no, please upload the crash dump file here so I can take a look at it. Thanks.

goonette, 2022-08-22:
you need to make sure Hyper-V is disabled in windows and that any virtualisation is disabled in your bios

cr4shboy, 2023-03-03:
Hey @Mattiwatti,

Same issue on Win11 22H2.
Pretty sure Hyper-V is disabled (checked, rebooted). Seems like boot-up process goes well, but when trying to use `EfiDSEFix.exe -d` after 2sec. it goes BSOD `Cl.dll - ATTEMPTED_WRITE_TO_READONLY_MEMORY`

Memory integrity, real time AV protection disabled.

[image]
[030323-5656-01.zip](https://github.com/Mattiwatti/EfiGuard/files/10882276/030323-5656-01.zip)

Mattiwatti, 2023-03-11:
Thanks for the crash dump. Can you also post the output of `EfiDSEFix.exe -i` please?

cr4shboy, 2023-03-11:
@Mattiwatti Of course.

```
SystemBootEnvironmentInformation:
        - BootIdentifier: {da4cf360-952e-11ed-ab5b-bfeced838031}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x2001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000081
        - HVCIOptions: 0x0000
```

Mattiwatti, 2023-03-11:
The `IntegrityOptions` in `SystemCodeIntegrityInformation` indicate that you do in fact have VBS running (`0x2000` = `CODEINTEGRITY_OPTION_HVCI_IUM_ENABLED`). `msinfo32.exe` should confirm this.

To disable VBS, try running `bcdedit.exe /set {current} hypervisorlaunchtype off`. Note that this will also disable Hyper-V (in case you would like to use it in the future I mean, since I see you already had it disabled).

I am planning to at least check for this flag in EfiDSEFix so as to not BSOD the system. I do have ideas for a more proper fix in EfiGuard itself but I can't say for sure yet if this will work out or not.

cr4shboy, 2023-03-11:
Indeed my bad,

It seems i checked, but you proved me wrong. Now worked like a charm!

Yeah, if there would be BSOD stopper and f.e. foolproof message, that VBS still enabled, it could stop the rookies like me, bugging you with such a questions.

Thank you so much @Mattiwatti and keep supporting this super tool!

commit 2f4a666b (2023-03-16): EfiGuardDxe: disable VBS for the current boot

This prevents a bugcheck on Windows 10 and later when VBS is enabled, which was made the default setting in Windows 11.
Additionally, EfiDSEFix will not proceed if it detects that VBS is still unexpectedly running (meaning either EfiGuardDxe was never loaded, or it failed to disable VBS).

Fixes #59
closed 2023-03-16 by Mattiwatti