Mattiwatti/EfiGuard #60: Failure: NetSystemEnvironmentValueEx error C000000D

LovelyDubby101, 2022-07-22:
Can someone please help me out with this issue thank you. 
I don't what is causing this but it gives me error C000000D.
Showing me
CI!g_CiOptions at 0xFFFFF802320393B8.
Failure: NtSetSystemEnvironmentValueEx error C000000D
AdjustCiOptions failed: C000000D

Mattiwatti, 2023-02-27:
Sorry for the late response. `C000000D` is `STATUS_INVALID_PARAMETER`, but I'm guessing that doesn't really help you.

Can you give some more information? What version of Windows are you using? Please also post the full output of `EfiDSEFix.exe -i`.

vedranbajic, 2023-03-15:
I am getting the same error. Important to note: I am using VeraCrypt Drive Encryption! Please let me know if that is an issue.

Here are my Infos:
`SystemBootEnvironmentInformation:
        - BootIdentifier: {009bd805-0e90-11ed-b455-daca6f0b8de9}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x2001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000`

Mattiwatti, 2023-03-16:
Thanks for posting the `-i` output.

> Edit 2: Alright. It is VeraCrypt's system drive encryption. I tried it on a VM without VeraCrypt and it worked JUST fine.

I think this might be a case of coincidence (meaning VeraCrypt is not related). Your output shows that VBS is enabled, which is known to cause issues.

Any chance you could retry with 2f4a666? See attached if you can't or don't want to compile the driver yourself, since there is no new release with this change yet: [EfiGuardDxe.zip](https://github.com/Mattiwatti/EfiGuard/files/10995223/EfiGuardDxe.zip)

Note that in general, `0xC000000D` may also be shown if the DXE driver was simply never loaded. While that is still basically a user error, I will be updating EfiDSEFix to show something more descriptive in this scenario since it is obviously not very helpful information on its own.

vedranbajic, 2023-03-16:
It is still not working for me. @LovelyDubby101 should try it again. 

My VeraCrypt encryption is getting in the way. I will create a new Issue regarding VeraCrypt and I am testing a possible solution! If that is out of scope — feel free to close it.

Mattiwatti, 2023-03-16:
Alright, thanks for checking.

> I will create a new Issue regarding VeraCrypt and I am testing a possible solution!

Great. If you can post steps for me to reproduce this on a (somewhat) minimal configuration, I will look into it and whether fixing it is feasible or not. And yeah, a new issue would be best.

vedranbajic, 2023-03-26:
A small update regarding the EfiGuardDxe.zip you provided:
I installed Windows 11, the most recent CU and downloaded your latest release from github. After that I got a "ATTEMPTING_TO_WRITE_READ_ONLY_MEMORY" bosd. However using the EfiGuardDxe in the zip file it worked perfectly fine.

Mattiwatti, 2023-03-26:
Hi,

Yes, that's correct, the latest release does not yet have the changes from 2f4a666 included, that's the reason I attached the driver separately as a zip file above. I am actually working on a new release though which should be out later today assuming all goes well.

closed 2023-10-04 by Mattiwatti