Mattiwatti/EfiGuard #63: Having a problem with Efiguard

Madfuckingman, 2022-08-20:
Hello. I'm trying to install Efiguard, but it doesn't boot at all (USB Stick). I've done all the steps but it just boots at windows, even putting to boot at my USB Stick on BIOS doens't work. I don't have neither Secure Boot or Hyper-V enabled. I'm using Windows 10 Education 21H2.

[image]

Mattiwatti, 2022-08-23:
As shown by the output of `EfiDSEFix -i`, your Windows installation is a legacy one ("FirmwareType: BIOS"). EfiGuard does not support non-UEFI installations of Windows.

closed 2022-08-23 by Mattiwatti

Madfuckingman, 2022-08-24:
I see, thanks for the help. I've used mbr2gpt command on my SSD and was able to change to UEFI mode.