Mattiwatti/EfiGuard #65: LoadImage Error 800000000000E (not found)

AdmiralNicherov, 2022-09-01:
I did it on youtube.
if you load as usual, this error occurs
[image]
here are the bios settings
[image]
[image]
and if you go into the bios without changing anything, click "save and exit", boot kit works successfully on Windows, gives an error
[image]
and more additional information
[image]

what could be the problem?

adzkii-i, 2022-09-01:
shouldn't the BIO be UEFI for EFIGUARD to work? ur bio is legacy

AdmiralNicherov, 2022-09-01:
> shouldn't the BIO be UEFI for EFIGUARD to work? ur bio is legacy

I have uefi enabled on the second photo you can see

AdmiralNicherov, 2022-09-03:
mbr2gpt finish problem

closed 2022-09-03 by AdmiralNicherov