Mattiwatti/EfiGuard #66: EAC and BE freeze my computer when I start a game and load the AC

NightfallChease, 2022-09-08:
EAC and BE freeze my computer when I start a game and load the AC

dotthefox, 2022-11-27:
I know this is like 2 months old but it ain't really worth using EFiGuard for bypassing EAC as EAC has recently made the bootkit detected and will refuse to launch the game

Sverys, 2023-02-06:
What would you suggest using? @dotthefox

ahmtergin, 2023-02-10:
Yes, I am kinda new to c++ as well, I don't know where to look at for an alternative? @dotthefox

Mattiwatti, 2023-02-27:
Hi,

You seem to be confused about the purpose of EfiGuard. EfiGuard is designed to disable PatchGuard, not to help you cheat in videogames. Note: if you think you need to disable PatchGuard in order to cheat in a game, you are probably wrong.

There are a number of ways (probably even very many) to detect or infer the absence of PatchGuard. And while I'm not claiming to know all of them, I do know of some. I haven't bothered to "fix" any of these (by making EfiGuard more like a rootkit) because... EfiGuard's job is not to help you cheat in videogames! Oh wait, I already said that. Closing this.

closed 2023-02-27 by Mattiwatti