Mattiwatti/EfiGuard #68: Can't install the efi

Fraudly, 2022-09-17:
I tried to install efi, but it says it does not support legacy (non-uefi) even though I have "UEFI only" in CSM
[image]

Mattiwatti, 2023-02-27:
From the device path being booted (`/USB(..),/HD(1,MBR,..)`) you can see that this is a legacy MBR drive (or possibly USB stick, whatever you are booting from). This is independent from the CSM or any other settings in the BIOS setup. A UEFI installation would show `/USB(..),/HD(1,GPT,..)` instead for a GPT disk.

Your BIOS is configured correctly, you simply need to partition the disk as GPT disk (a Google search should point you to some guides for how to do this). EfiGuard is working as intended in this scenario since it does not work with legacy (MBR/BIOS) installations of Windows like it says on the screen.

closed 2023-02-27 by Mattiwatti

Fraudly, 2023-11-07:
Hi again, haven't posted in a while, just forgot about it, but after a while I tried to run efi again, even bought a new flash drive and with rufus made the partition as GPT, but I didn't understand why the installation didn't
 go at all, there wasn't even any error at all.
[image]