Mattiwatti/EfiGuard #72: Booting causes BSOD

devingDev, 2022-12-23:
Trying to boot with the 1.2.1
it fails and causes a BSOD with some thread exception or whatever
on the bsod it also displays four hex codes on the top left.

SystemBootEnvironmentInformation:
        - BootIdentifier: {7e94e7d5-7e54-11ec-8e89-e7fba18db13c}
        - FirmwareType: UEFI
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x2001

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 0
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 0
        - DebuggerEnabled: 0
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x00

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000

stevefan1999-personal, 2023-01-09:
Seems like some Windows internal structures are changed...

Mattiwatti, 2023-02-27:
Hi,

Thanks for creating an issue, but if you want me to fix a blue screen I'm going to need something more to go on than
> a BSOD with some thread exception or whatever

and
> it also displays four hex codes on the top left

Preferably I need a crash dump (a 'kernel' or 'automatic' dump will do), but at the *very* least I need to know the bugcheck code. What does the screen actually say? That includes the four hex codes since they are the bugcheck parameters. Still, without a crash dump it's not likely I will be able to fix this unless the issue is very obvious. So, need more information.

Mattiwatti, 2023-02-27:
Backpedaling because I missed this:
> IntegrityOptions: 0x2001

I'm assuming you are on Windows 11, or else 10 with VBS enabled. Try `bcdedit /set hypervisorlaunchtype off` followed by a reboot. If that's not it, see previous message.