Mattiwatti/EfiGuard #73: Bsod if virtualization technology is turn on in bios settings

Crazycat070, 2022-12-30:
I get a Blue Screen of Death when i type EfiDSEFix.exe -d in command prompt. This is only after i type it the computer boots up fine until the command is typed. The command works fine if virtualization technology is turned off but for my use case i need to have it on.

Matty66666, 2022-12-30:
by chance do you have secure boot enabled in the bios? Also maybe verify that your virtualization enabled in the bios is not HVCI 
 or even worse where you have HVCI, VBS, and already set TPM keys. option, I venture you are looking for VT-x, VT-d  (assumming you are an Intel fan) type of hardware support such that nested page tables are fully in your control (ignore my ignorance on my last statement - NPT is only a part of the solution assumming your BIOS is correctly confgured.

Let me offer some other possible avenues I have found to be useful in the past:

- Windows 11 22H2 has openly available doctored distributions which remove much (dare I say more than most) of the Microsoft annoyances. However, this approach comes with problems. For example OS stability, parnoid thoughts that MS doesn't own your computer anymore but the developers that provided the manipulated Win 22H2.
- You are in control of your own computer! Fundamentally (PLEASE CORRECT ME IF I AM WRONG ON THIS POINT!) - for example I thought that one can self-set their own TPM keys in the bios which would then (PLEASE CORRECT ME IF I AM MISTAKEN) allow the owner of the computer to essentially set up an environment where they are able to self-sign their own drivers without messing around with trrying to counterfiet or trick windows into accepting your kernel driver. Fundamentally I thought (I need to read more!) that if you own your own computer and you have the know-how then you do not have to worry about testsigning flag nor even enablng debug mode nor enabling OEM or PE mode booting which is what I thought OEMs have to do to install all of their spyware before selliing to customers.
- Another possability is to say screw it - utilize KDU utility and boot via EFIGuard (use the safer method between the two on bootup). If EFIGuard boots then now you know that Patchguard is inert (afterall EFIGuard was written by Matti so that is a fairly safe bet). Once patchguard is out of the way then I guess you don't even need KDU (a tool that is able to place unsigned driver code into a protected Win11 installation (last I played wth that tool it worked amazing!). 
-Another option is to use nested virtualization and hypervisor introspection to accomplish whatever you are trying to do under a fully patched and patchguard protected kernel. Just place your development insidee of say VirtualBox (needs to be configured correctly as to avoid common anti-VM techniques - perhaps look into hardening your virtualbox, xen, vmware, hyper-v, ... If I reacall correctly you might need ot edit config variables related to time sync between the host and the VM as to avoid ~costly in accommulation heuristics that are used to detect your VM which is the opposite of the objective.
--There are various relatively new github projects that are dedicated to specifically allowing the user to succesfully employ 'stealthly' runtime hooking of the guest VM and other neat functions that come in handy. This approach may be the easiest for you if you have a Intel processor and are willing to relax requirements on the speed and core-count of the VM you are expermenting with. Also this route is only a github search away. Check out ~DVAWF or is it DWARK ... There are a variety of them now open-source. I personally have had great success in the past with IceBox introspection and their extremely fast IPC created by modification of the virtualbox  kernel drivers (caveat - these drivers will be run outside of your safe VM so simply doing something like bcdedit /set testsigning on in a windows shell on the host I found to be more than adaquate for my issues then.
-Another option is to reverse engineering Windows Kernel Explorer and duplicate how the developer/developers on that project accomplish what I have been ranting about without all the fuss for the end user (I am >90% sure that those devs do not have MS approved cerificate backing their kernel code nor do I know much about how they get away with that considering MS and countermeasures (i.e., unsigned kernel code running inside a ~secure Win11 22H2 kernel. Maybe Matti might have some ideas on that (not maybe... I would bet $1,000 he/she does know. And Matti is extremely helpful when posed with a reasonbly phrased and coherent question that is too the point. But of course Matti is Matti an not me so I cannot speak for him/her). Get on discord, talk to people, experiment on your own, fail ALOT and hopefully your nominal failure rate will take a step in positiivty.

Apologies for long response. I hope it wasn't a complete waste of time for anyone who read it.

Other options are available - some hard and some easy. An easy one is to go back to Windows 10. A hard one is to start your attack unconventially such as embedding yourself in the SMM layer and then meticulally working your way from the lower level to the upper levels such as wiggling your way from SMM to the ~signed and protected poritions of the MS kernel. Once there how hard could be really be to then wiggle your way to userland while maintining your previous persistence in those lower layers. (I am mostly kiding because I do not know how to do that... I have heard of supply chain attacks recently that however were claimed to perform such a sophiscated and target attack).

I am sure others here will think me a fool and I hope they respond with a simplier solution than the things I have proposed which I wouldnt call easy nor would i call hard (except that last technique I mentoned). That way I can learn something new that works better than the things I have figured out in some part. I am still searching for easier and simplier ideas/solutions. I am a bit concerned about VBS + HVCI + other hardware and software backed mitigations to me not having control over my own paid for computer and operating system. Linux is lookng more like the pancea each passing day.

Best, 
-skinnyM

closed 2023-01-04 by Crazycat070

Mattiwatti, 2023-03-16:
This was fixed in 2f4a666.