Mattiwatti/EfiGuard #74: Can’t remove Efiguard

charlesknapp, 2023-01-12:
Hello,
After quite some time of installing Efiguard, my PC decided to update (I believe) and there is an issue with missing files (\windows\system32\winload.efi and winload.exe)

how would I go about repairing this? I can’t boot from any USBs or repair my PC with an installation media USB.

I just want to remove this bootkit from starting every single time I boot my PC. It seems to try and patch when I run my installation media disk, and just shows the same missing file error each time.

Thanks for your assistance!

matt-src, 2023-02-02:
@charlesknapp did you find any solution?

Mattiwatti, 2023-02-08:
What makes you think the issue is a **missing** `winload.efi`?

Given the timeline of events (as given by you) is...
1. You installed EfiGuard.
2. Some time after this, you ran Windows Update.
3. Windows broke. (present situation)

....which do you think is more likely to have caused the problem: EfiGuard or Windows Update?

EfiGuard doesn't modify files on disk, nevermind delete them. So I'm going to go with no, this is not an EfiGuard issue.

It is of course possible that you want to stop using EfiGuard regardless. If so, you can follow the following guide to uninstall it:

1. Delete `EfiGuardDxe.efi` and optinally `Loader.efi` if you were using it.
2. EfiGuard is now uninstalled.

closed 2023-02-08 by Mattiwatti