Mattiwatti/EfiGuard #75: Enter "EfiDSEFix.exe -d" but no response on win10 1909 18363.2158

Emiya0415, 2023-01-26:
Sorry for my poor English, but I have done all things follow video on youtube and I saw the green successful message after launching by uefi when start windows.
[https://www.youtube.com/watch?v=_auePp1nTHs](url)
I enter the commad "EfiDSEFix.exe -d" on right folder and also with admin right, but it don't response at all
I don't know what to do next and please help me 
[image]
[image]

Mattiwatti, 2023-02-27:
I'm not sure what the cause of this would be, but a wild guess is that it has something to do with your Windows being set to Chinese. Which locale/culture are you using? Simplified Chinese (zh-CN)?

wawaovo, 2023-04-19:
> 

I got this problem too , only the first time I boot my device by this method and the first time I enter the commad "EfiDSEFix.exe -d" it work and I try to enter the commend "EfiDSEFix.exe -e" it just response at all.  After that I try to do the process again , and this time even the commad "EfiDSEFix.exe -d" wont work.

ashyerv, 2023-04-27:
@Mattiwatti Mattiwatti
use `zh-CN` `Windows 101909  (18363.2158)`
shell use `cmd.exe`

wawaovo, 2023-04-27:
@Mattiwatti I use zh-TW windows 10 19045.2946 shell use cmd.exe

Mattiwatti, 2023-05-11:
Thanks. I just tested this using Windows 10 and I can't reproduce this here using zh-TW:

[image]
Same for zh-CN.

I'm not sure what to do about this since I can't debug the issue if I can't reproduce it, but here are some more questions:
1. Does `EfiDSEFix.exe -i` also exit with no output?
2. What is the result of `echo %ERRORLEVEL%` from a `cmd.exe` prompt after running `EfiDSEFix.exe -d`? This is the program exit code.
3. Can you try the attached version of EfiDSEFix instead and tell me if you get output? This version uses the CRT for printing. I don't know if this will make a difference, but other than this I don't have any ideas.
  [EfiDSEFix-v1.3.1-CRT.zip](https://github.com/Mattiwatti/EfiGuard/files/11456688/EfiDSEFix-v1.3.1-CRT.zip)

Mafferis, 2023-05-14:
I had the same problem and i figured it out that you need to run cmd as admin and it will work.

Mattiwatti, 2023-05-17:
Thanks! You're right, I was able to reproduce this (regardless of UI language - this was a red herring) by enabling UAC.

Unfortunately this is not so easy to fix, due to the following:
1. `EfiDSEFix` needs to run as administrator because this is required for the syscall it makes in `-d`/`--disable` mode (some other modes like `-i` could technically run without elevation, but `-d` is really the reason the tool exists in the first place).
2. Windows' handling of programs requiring administrator access is subtly broken for console windows, in that Windows creates a new process, with a new window. (To be clear: the 'new process' is new even accounting for the new process which you are of course creating - in other words, the program runs twice.)

Because of (1), there is an embedded manifest in `EfiDSEFix.exe` which ensures that the program runs as administrator. If this is not possible, it does not run at all. So far so good. But because of (2), if you are in a `cmd.exe` session that is *not* elevated, and you run `EfiDSEFix.exe`, Windows will create the elevated process with a different (hidden) console window from the one you are using for the `cmd.exe` session. Therefore **EfiDSEFix is actually (probably) working**, but you can't see what it is printing.

Ironically it is the manifest that is causing this issue when UAC is enabled: removing it (or changing its `requestedExecutionLevel` to the default of `asInvoker`) makes printing to the console work. But then of course the program does not actually work due to (1), so this is not really a solution.

I have some ideas about how to deal with this, but they are not great due to requiring large amounts of code and/or having other undesirable side effects like creating a new console window or waiting for a keypress. While this is a real bug and I'm leaving it open as such, it's still relatively unimportant and I'm not willing to 'fix' it unless I can find something that does not degrade the user experience in other ways. @Mafferis has already given the best workaround which is to simply run `cmd.exe` as administrator and then run `EfiDSEFix.exe`.

Mattiwatti, 2023-10-14:
This was fixed in f04570f, which will be in the v1.4 release.

The fix I ended up going with changes the `requestedExecutionLevel` to `asInvoker`. Therefore `EfiDSEFix` will no longer automatically prompt for elevation if UAC is enabled, and running it from an unelevated prompt will simply fail. I've changed the error handling for this to explain that the program must be run as administrator, and then quit.

So this is somewhat of a low tech solution, but I think this is ultimately better than the invisible console situation described above. I tried many other approaches to solving this, but all of these had some other issue(s) or regressions in other areas that were not acceptable.

Fun fact: Microsoft's own console programs suffer from this too! Try `WinSAT.exe /?` from an unelevated prompt to (not) see.

commit f04570f7 (2023-10-14): EfiDSEFix: do not request UAC elevation in manifest

This works around a Windows Shell issue when launching console applications that require admin elevation (this includes EfiDSEFix) from an unelevated command prompt while having UAC enabled.

Fixes #75
closed 2023-10-14 by Mattiwatti