Mattiwatti/EfiGuard #76: EAC Patched Windows boot loader detected. (Kernel Patch Protection disabled)

Pain160710, 2023-02-18:
When try to start The Division 2 which is protected by EAC, i get this message after using EfiGuard.

1. I`ve booted from the USB-Stick with the EfiGuard files
2. I`ve opened the cmd as admin, navigated to the save directory of EfiGuard and entered this command: Efidsefix.exe -d
3. After that i started modded CheatEngine without a problem
4. After that i entered this command in the same cmd window: Efidsefix.exe -e
5. Finally i started the game and got the message: "Patched Windows boot loader detected. (Kernel Patch Protection disabled)"
Do you have any workaround for this problem?

Thank you and have a nice evening, best regards.
[image]

DecoderCoder, 2023-02-22:
Can you check other game?
And by the way, check this https://github.com/ByteWhite1x1/PatchGuardResearch

Mattiwatti, 2023-02-27:
See #66.

closed 2023-02-27 by Mattiwatti