Mattiwatti/EfiGuard #80: Error when using certain applications

KingOwnageXV, 2023-03-29:
I saw Mattiwatti's comments but he locked the threads after, I wanted to alpine on this as it is an unresolved issue and his comments don't accurately describe the reasons why every single person may want this

a) need this disabled for other purposes and don't feel like rebooting just to swap to a game (me)

b) can't do my work and play games at the same time (me)

c) some anti-cheats can harm performance

d) anti-cheats are a huge privacy invasion due to all the data they collect and the ToS of the games stating how they will use that data, as they have so much access to your PC. People who value privacy do not not like this

I don't know what the other people's intent are, but equating bypassing AC with cheating isn't necessarily true, and in all honesty there are much much better ways to do so to begin with so I don't feel like they're trying to cheat or if they are they're tech illiterate and will just get banned if they do decide to inject a probably detected dll or access memory as being able to run unsigned drivers isn't the end-all-be-all of anti-cheat, you'll still get caught it requires more than that. 

Personally I'm in group A and B, I use this for security and convenience being able to enable and disable unsigned drivers because I'm a developer for Reddit, I work long and I play games on the side. I've been having to choose between convenience and entertainment which is annoying. Anyway I do not expect any sort of update to this program to be made to accommodate my niche issue (unless other users who posted had the same reasons, but even then) I just wanted to let you know we're not all cheaters and do what you want with that information, but hopefully this is the last post about the "issue". Cheers mate

Mattiwatti, 2023-03-29:
Dear KingOwnageXV,

I think you'll find you're wrong: I've never locked a single issue on Github. A *closed* issue is not the same thing as a locked issue, and an issue can be closed without being locked and vice versa. You can continue to post in a closed issue, and me and everyone else who has notifications for issues enabled will still be notified by email when you do so (you may want to keep this in mind for the future). If I close an issue, that simply means I consider it to be resolved, which in turn is not necessarily the same thing as being fixed! After all, `RESOLVED WONTFIX` is the most satisfying kind of `RESOLVED`.

Other than this, I'm not sure why you thought that the best way to get me to fix your issue was to create a new issue with more words and less information than e.g. #66. I didn't even make it to the third paragraph! Since this is not a new issue but rather an overly wordy recreation of an existing one, I'm going to close this as well (bet you saw that coming).

Things I need in order to take an issue seriously:
- A title that describes the specific issue (not: "error when using certain applications")
- A brief description of the issue (not: vague generic references to "Matti's comments" and "the threads" which leave me guessing as to what you are even talking about)
- How to reproduce the issue

Things I don't need, and probably don't want, in any issue:
- Essays about your life as a Reddit moderator

Your essay, while touching, really belongs in #66, with something like "Matti, [please] consider reopening this issue for the following reasons: <list>". And then I will! Consider it, that is.

Note that checking the first three boxes above simply means I'm not going to close the issue outright, not that I'm going to fix it or even look into it - especially when the steps to reproduce presumably involve buying some game. You are of course free to make a PR that fixes the issue.

closed 2023-03-29 by Mattiwatti

KingOwnageXV, 2023-03-30:
> Dear KingOwnageXV,
> 
> I think you'll find you're wrong: I've never locked a single issue on Github. A _closed_ issue is not the same thing as a locked issue, and an issue can be closed without being locked and vice versa. You can continue to post in a closed issue, and me and everyone else who has notifications for issues enabled will still be notified by email when you do so (you may want to keep this in mind for the future). If I close an issue, that simply means I consider it to be resolved, which in turn is not necessarily the same thing as being fixed! After all, `RESOLVED WONTFIX` is the most satisfying kind of `RESOLVED`.
> 
> Other than this, I'm not sure why you thought that the best way to get me to fix your issue was to create a new issue with more words and less information than e.g. #66. I didn't even make it to the third paragraph! Since this is not a new issue but rather an overly wordy recreation of an existing one, I'm going to close this as well (bet you saw that coming).
> 
> Things I need in order to take an issue seriously:
> 
> * A title that describes the specific issue (not: "error when using certain applications")
> * A brief description of the issue (not: vague generic references to "Matti's comments" and "the threads" which leave me guessing as to what you are even talking about)
> * How to reproduce the issue
> 
> Things I don't need, and probably don't want, in any issue:
> 
> * Essays about your life as a Reddit moderator
> 
> Your essay, while touching, really belongs in #66, with something like "Matti, [please] consider reopening this issue for the following reasons: ". And then I will! Consider it, that is.
> 
> Note that checking the first three boxes above simply means I'm not going to close the issue outright, not that I'm going to fix it or even look into it - especially when the steps to reproduce presumably involve buying some game. You are of course free to make a PR that fixes the issue.

Thanks for informing me that a closed issue is not the same as a lock issue, stating you locked threads was just my way of explaining why I was creating a report for something previously mentioned not to diss you, my entire post was polite.

Also not only was my post not ride but your reply is also very long just like mine was, yet you stated you didn't finish reading it. You've just done everything you accused me of doing and you also misinterpreted my intentions probably because you didn't finish