Mattiwatti/EfiGuard #81: Request: Feature: Quiet Mode

SomeDude3, 2023-04-02:
I wonder if it is possible for a quiet version of this bootloader? Allowing to show the bootscreen/Logo throughout the boot process? My BIOS supports enabling and disabling the BGRT, and when the BGRT is disabled, Windows loads bootres.dll to make use of the boot logo. With this bootloader, it is actually possible to use a customized version of bootres.dll, however the image does not last long, due to the text outputs.

Mattiwatti, 2023-04-05:
This should be doable but it is not a very high priority for me. I will put it on the to do list for the next release (if/when).

Mattiwatti, 2023-10-04:
Having second thoughts about my commitment to adding this. The main reason for this being that I honestly don't really want to add more configuration parameters to the driver (in fact I would prefer to get rid of `Loader.config.efi` entirely), and I **definitely** don't want to add a second variant   of the loader (so meaning something like a `Loader.quiet.efi`).

Also see PR #101, which had similar issues, where I explain my reasoning for above in a bit more detail.

A second, less important, reason would be facilitating development of "stealth" malware based on EfiGuard. I know this isn't the strongest ever argument (by this logic, why even create a bootkit - making a stealthier EfiGuard probably isn't exactly the main thing holding back advanced UEFI malware development at this point), but I have to draw the line somewhere and adding a "quiet mode" is it.

---

That said (and closing this issue), I had a look to see what actually implementing this would entail, and you probably won't be surprised that it isn't very much. Adding the following to `EfiGuardDxe/util.h` will take care of most print statements:
```c
inline
UINTN
EFIAPI
NullPrint(
	IN CONST CHAR16 *Format,
	...
	)
{
	(VOID)Format;
	return 0;
}

#define Print(Format, ...) NullPrint(Format, ## __VA_ARGS__)
```

After this there will still be some print statements remaining because the literal `Print()` function cannot/is not called everywhere; in some places EfiGuard uses `gST->ConOut` directly for various reasons (including printing but also sometimes for clearing the text buffer or changing the text foreground/background colours). If you grep for `"gST->ConOut"`, I expect getting rid of these should be straightforward enough. If you need help with the C side of this, feel free to make a fork and I can create a pull request to it with the changes.

Basically the TLDR of this is that I don't want to do this feature not for implementation reasons but because I don't want to maintain and support it. Do let me know if the above works for you (or else any problems/questions) though as I actually think your use case is quite interesting.

closed 2023-10-04 by Mattiwatti