Mattiwatti/EfiGuard #85: UefiSeven coexistance

acoul, 2023-05-02:
Greetings,

I use [UefiSeven](https://github.com/manatails/uefiseven) to boot Windows 7 on a UEFI only system. Trying to test EfiGuard from a usb flash, I get an error: can't find bootmgfw.efi version number.

can EfiGuard coexist with UefiSeven ? if not, how Windows 7 can use EfiGuard on a UEFI only system ?

Mattiwatti, 2023-05-03:
Hi,

I took a look at UefiSeven and there's no reason why it wouldn't *theoretically* work with EfiGuard. Unfortunately however because it replaces the Windows boot manager file (`bootmgfw.efi`) on disk, EfiGuard won't work with it as is because EfiGuard relies on this being the original Windows boot manager. This explains the warning you are seeing (not being able to find the version number is only a warning - it isn't fatal), as well as any subsequent errors which you will surely also receive if the Windows boot manager is located elsewhere with a different name.

So in short, no, EfiGuard won't work with UefiSeven as is, though there is no reason why it couldn't work if UefiSeven used a different method of loading.

> if not, how can Windows 7 use EfiGuard on a UEFI only system ?

See wiki: [Secure boot on Windows 7](https://github.com/Mattiwatti/EfiGuard/wiki/Secure-boot-on-Windows-7) (if you don't need secure boot enabled, you can skip the last section, and disabling the CSM is also not necessary in this case).
In short: you'll want to have up-to-date graphics drivers installed or slipstreamed on Windows 7, and the boot manager updated to one from Windows 10 or 11. Then disable the `Vga` and `VgaSave` services, and run `bcdedit /set {current} novesa on` and `bcdedit /set {current} quietboot on`.
Note: all of the steps are detailed the way you would run them from Windows 7 itself, but they can also be performed externally, e.g. from Windows 10/11 or a bootable ISO of same.

acoul, 2023-05-04:
you are quite correct about the UefiSeven. it's not an error, just a non destructive warning. EfiGuard seems to overcome this and proceeds successfully on each patching stage. I switched from Loader.config.efi back to the default Loader.efi & all the steps (with the exception of the bootmgfw.efi version warning) went just fine.

unfortunately, the result is a gabbled half screen output. I can kind of see the desktop with menus etc. but the result looks like a bad acid trip, which BTW, I've never tried so far. fortunately, VNC, that I use for remote access, works fine. 

Now I can try to install the graphics driver (Intel UHD 600 9th Gen) & upon success, test booting Windows 7 in pure UEFI by directly using the graphics driver skipping the generic [Vga and VgaSave services](https://github.com/Mattiwatti/EfiGuard/wiki/Secure-boot-on-Windows-7#upgrading-the-boot-manager) as you suggested & have a follow-up.

acoul, 2023-05-11:
I am proceeding with the [Vga and VgaSave services](https://github.com/Mattiwatti/EfiGuard/wiki/Secure-boot-on-Windows-7#upgrading-the-boot-manager) guide.

I have installed the graphics driver, booted Sergei Strelec's WinPE, renamed vga.sys & vgapnp.sys, used build-in remote regedit & switched the Start value to 4 according to the guide.

but the bcdedit commands through the following error:

`The boot configuration data store could not be opened.`
`The requested system device cannot be found.`

Mattiwatti, 2023-05-11:
Are you sure Windows (so the partition containing the `Windows` directory) is on `C:` here if the BCD store (so normally the ESP) is too? I think this is probably possible on most systems nowadays, it's just not common to see this so just making sure.

Since you are running `bcdedit` from WinPE, you'll need to specify the BCD store location when running `/set` if you weren't doing this already, so just as you are doing with `/enum` above. You'll also have to specify the GUID explicitly, `{current}` won't work.

So the commands become (for the first entry):
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {default} novesa on`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {default} quietboot on`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {default} bootstatuspolicy IgnoreAllFailures`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {default} bootlog yes`

Repeat for `{12a3b634-f027-11ed-aee3-18c04dd05a2e}` and `{8d481ea9-efcf-11ed-8773-a8a1594ec55d}` if you want to do this for all three entries.

Note: your other two entries have a missing device for the Windows partition (I'm guessing from the conversion to GPT), if you want to boot these you'll also need to run (assuming Windows is on `C:`):
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {12a3b634-f027-11ed-aee3-18c04dd05a2e} device partition=C:`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {12a3b634-f027-11ed-aee3-18c04dd05a2e} osdevice partition=C:`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {8d481ea9-efcf-11ed-8773-a8a1594ec55d} device partition=C:`
`bcdedit /store C:\EFI\Microsoft\Boot\BCD /set {8d481ea9-efcf-11ed-8773-a8a1594ec55d} osdevice partition=C:`

acoul, 2023-05-14:
I am still working on this & hopefully have a detailed followup soon

Mattiwatti, 2023-05-17:
This looks like a firmware bug unfortunately. After testing this on a Ryzen system I arrive at the same conclusion (no VGA output without CSM), even though the same setup works on an older Intel system. I suspect this is somehow related to differences between their shared vendor firmware code (usually provided by AMI), or even caused by the video driver differentiating, since it does not make sense for the processor to be a factor in this. But this is just speculation.

I will close this issue now since it's not really a bug in EfiGuard and I can't help you further with it. However I'd be interested to hear about it if you do find a fix for this, since (depending on the fix) it might be possible to incorporate it into EfiGuard.

closed 2023-05-17 by Mattiwatti

acoul, 2023-05-17:
It is my Bad that I mixed couple of different hardware in-order to further explore the  EfiGuard functionality.

The initial bug report has to do with an Intel J4125 UEFI Class 3 system that UefiSeven manages to provide VGA service that boots successfully to Windows 7. On this system, I am unable to install an unsigned graphics driver (UHD 600), thereof I wanted to try this procedure with EfiGuard (this is still a WiP for me)

Using EfiGuard along with UefiSeven results in an garbled video output similar to the following:

[image]

 This is the actual (still active) BUG that I wanted to report initially, find where the problem resides & perhaps resolve it ( besides my own quest to successfully install the graphics driver under the UEFI Class 3 system).

the rest, is some of my (off-topic) attempts to disable the VGA driver (as you kindly suggested) & try to UEFI boot directly to the graphics driver, in this case a Radeon HD 8490

Should this ticket perhaps be split to two ? the original issue (still active) and the later one (bypassing VGA driver) ?

acoul, 2023-05-26:
This is my final report on this issue

Intel UHD600 is a tempered graphics unit. I was unable to find any working drivers for either win7 nor win8

for win7 I had to use UefiSeven, as my j4125/UHD600 is UEFI class 3 system, meaning no CSM capability 

* Under win7:
UefiSeven works fine with UHD600 but is stuck at resolution 1024x768 since there are no working drivers as UefiSeven locks the resolution at 1024x768. EfiGuard in combination with UefiSeven results on a garbled video output (the reason for this is still unknown) 

* Under win8:
no UefiSeven is needed, but since there are no drivers available either, resolution of the MS Basic display driver is stuck at 1024x768. Surprisingly, booting with EfiGuard the resolution is unlocked to the full potential of the display monitor. the other positive news, in this case, is that the MS Basic display driver offers full DirectX 11 acceleration according to dxdiag

[image]

* Under win10: no need (yet) to use EfiGuard, but the later it does work fluently

in all cases, EfiGuard worked wonderfully !

I used [bootice](https://www.majorgeeks.com/files/details/bootice_64_bit.html) to embed the EfiGuard onto the UEFI menu under the EFI\Boot directory.