Mattiwatti/EfiGuard #86: Question

NightRaidGamer, 2023-05-09:
hello I would like to know if this bootkit supports old systems that are not compatible with UEFI?

Mattiwatti, 2023-05-09:
Nope. Sorry.

closed 2023-05-09 by Mattiwatti

NightRaidGamer, 2023-05-09:
> Nope. Sorry.

could you add support in the future?

Mattiwatti, 2023-05-10:
No, not really. EfiGuard relies on a number of things present in UEFI (like EFI runtime services) which do not exist in BIOS - this the reason why it's named EfiGuard.

NightRaidGamer, 2023-05-10:
> No, not really. EfiGuard relies on a number of things present in UEFI (like EFI runtime services) which do not exist in BIOS - this the reason why it's named EfiGuard.

what about MBR and VBR ?

Mattiwatti, 2023-10-04:
What about them? Those are just ways (well, places) to gain code execution, which is not the issue.