Mattiwatti/EfiGuard #88: How to use EfiGuard with Hibernation on windows 10, if its supported?

burritosilike, 2023-05-18:
I tried to make my USB to be the first boot but the windows always boots up freshly, as if I did a shutdown / restart.

Is it possible to use hibernate along with it?

Thank you!

Mattiwatti, 2023-05-23:
Hi,

Does EfiGuard work with ACPI S4/hibernate? Yes and no (mostly no - but depending on what you want to do precisely).

Yes, you can use EfiGuard together with ACPI S4, but the DXE driver will not be reloaded when resuming from S4. This means that in the default boot configuration, i.e. using the EFI runtime `SetVariable()` hook, the driver can only be used to write to `g_CiOptions` after a full boot. This is because the boot entries in the BIOS (so in this case, your USB stick containing `Loader.efi`) are only processed during boot, and not when resuming from S4. I did look into some ways to get this to work while finding the answer to this question for you (most notably using `bcfg driver addp` to add `EfiGuardDxe.efi` as a driver option), but none of these seem to work reliably (if at all) so far from what I can tell.

PatchGuard will always be disabled and stay disabled.

So what does this mean? If you don't need the ability to toggle DSE on and off, or disabling it once and leaving it disabled is fine, EfiGuard will work fine (assuming you follow the instructions in the next paragraph to work around the issue you are seeing). In this case it's probably easiest to use `Loader.config.efi` to tell the driver to disable DSE permanently at boot time instead of using the runtime services hook. If you *do* need the ability to toggle DSE on and off, EfiGuard won't work for this after the first resume from S4 due to the issue I described above.

Re: the issue you are seeing (resume from hibernate doing a full boot instead of returning to Windows): this is happening because EfiGuard is the first entry in the boot entry list. While this works fine when booting, it breaks resume from S4. There are two workarounds for this, which both pretty much accomplish the same thing. Either:
1. Configure your boot entries in the BIOS so that "Windows Boot Manager" is the first boot entry, not EfiGuard. Put EfiGuard anywhere else in the list (but don't delete it).
2. After starting the computer, press the hotkey for the boot selection menu (which key this is depends on your motherboard - try F8, F10, F11 or F12). Then choose EfiGuard and boot from it.

*or:*

1. Configure your boot entries in the BIOS so that EfiGuard is first, and "Windows Boot Manager" is second (i.e. the way you've got it now).
2. Start the computer and let it boot into Windows after loading EfiGuard.
3. Use an external utility such as [BootICE](https://bootice.en.softonic.com) or [EasyUEFI](https://easyuefi.en.softonic.com) to change the boot entries so that "Windows Boot Manager" is now first in the list.

(As you can see, the first method is much simpler, so try that one if possible. If your motherboard hasn't got a boot selection menu, try the second method.)

Both of these will make it so that Windows is once again the first entry in the boot entry list, and resuming from S4 should once again work as expected after doing this.

burritosilike, 2023-05-24:
Ah, yes, I do need the ability to toggle DSE on and off.

If that's the case, that's all right. 

Thanks again for the time and effort in answering it!

Feel free to close this issue (I'm not sure which close it will be).

closed 2023-05-24 by Mattiwatti

Mattiwatti, 2023-05-25:
Seems I closed this prematurely: I just managed to get this to work (meaning ACPI S4 using the default configuration which toggles DSE on and off) using `bcfg driver`. I wasn't expecting this to work since I originally couldn't even get this to work with regular boots outside of a VM, let alone with ACPI S4 resume.

You'll probably want to have `EfiGuardDxe.efi` on the ESP for this (so not a removable drive), though it shouldn't matter in principle so long as the drive has the same UEFI device path every time. Here's what I did:

1. Remove the boot entry for EfiGuard if you have one, and make sure "Windows Boot Manager" is at the top of the list. ACPI S4 sleep/resume should be working at this point (sans EfiGuard of course).
2. Boot into the UEFI shell and `cd` to the drive and directory containing `EfiGuardDxe.efi`, e.g. `fs0:\EFI\Boot`.
3. Run `bcfg driver add 0 EfiGuardDxe.efi "EfiGuardDxe"` followed by `reset`.

At this point you will now either see EfiGuard messages during boot, or no change (i.e. the driver is not being loaded - verify this with `EfiDSEFix.exe -c`). If the former, the driver should now be reloaded after every resume from S4, and you are done. If the latter, repeat steps 2 and 3 above but using **addp** instead of **add** in step 3. In my testing it seems to vary which of these you need to use; I had to use `addp` in a VM with OVMF, and `add` on a bare metal Asus desktop motherboard.

Note: depending on your firmware there may be a third possibility, namely that neither of these works. If this is the case I can't help with that. I haven't actually seen this be the case though.