Mattiwatti/EfiGuard #89: Error: AdjustCiOptions failed: 0xC0000002

Jack3De, 2023-05-20:
I'm using version 1.3. I am getting an error:
Disable DSE...
CI!g_CiOptions at 0xFFFFF8011D829418.
Failure: NtSetSystemEnvironmentValueEx error 0xC0000002
The EfiGuard DXE driver is either not loaded in SETVARIABLE_HOOK mode, or it is malfunctioning.
AdjustCiOptions failed: 0xC0000002

I renamed the file on the flash drive, deleted the config, the flash drive was formatted in FAT, the file launch from the flash drive into the BIOS and Windows is successful.
[image]

[image]

Mattiwatti, 2023-05-21:
Hi, there's really no need to email me to tell me to take a look at your issue less than an hour after creating it, GitHub already does this for you.

Code `0xC0000002` is `STATUS_NOT_IMPLEMENTED`. The HAL returns this when there are no EFI runtime services available for whatever reason. What this reason is, I can't say, but most likely this is somehow being caused by your system firmware (Dell BIOS).

Questions:
- Can you give the output of `bcdedit /enum all`? (attach as a text file since the output will probably be very long).
- Same question for `bcfg boot dump -v` (UEFI shell) or `efibootmgr -v` (Linux). One is enough, these show the same information.
- Can you try EfiGuard on another system (preferably not Dell), and if so, does this work?
- Is Secure Boot enabled in the BIOS?

Note that I doubt that I can fix this (assuming I'm right about the cause being the system firmware), but I may be able to identify what the cause is more precisely. As of right now I don't really have any ideas about what this could be.

Jack3De, 2023-05-21:
1) [bcdedit.txt](https://github.com/Mattiwatti/EfiGuard/files/11525242/bcdedit.txt)

2) Of the three different computers, on one I managed to do this. Lenovo ThinkBook. On another laptop, HP, I got a different error, 0x000000E

3) Secure Boot disabled

Jack3De, 2023-05-22:
Any ideas how to resolve this issue?

Mattiwatti, 2023-05-25:
Nope, sorry. This does seem to be some hardware (or rather, firmware) specific issue, so there's not much I can do since I haven't got this specific laptop model to reproduce this.

I'll leave the issue open in case anyone has suggestions.

horsicq, 2023-08-19:
The same error:

C:\ownCloud\progs_win\EfiGuard-v1.3>EfiDSEFix.exe --disable
Disabling DSE...
CI!g_CiOptions at 0xFFFFF8010FA393D8.
Failure: NtSetSystemEnvironmentValueEx error 0xC0000002
The EfiGuard DXE driver is either not loaded in SETVARIABLE_HOOK mode, or it is malfunctioning.
AdjustCiOptions failed: 0xC0000002

Mini PC https://www.gigabyte.com/Mini-PcBarebone/GB-BACE-3000-rev-10#ov

horsicq, 2023-08-19:
C:\ownCloud\progs_win\EfiGuard-v1.3>EfiDSEFix.exe -i
SystemBootEnvironmentInformation:
        - BootIdentifier: {66e17ee1-29d7-11ee-901d-f2c7d7eecd99}
        - FirmwareType: BIOS
        - BootFlags: 0x0

SystemModuleInformation:
        - Kernel: ntoskrnl.exe (\SystemRoot\system32\ntoskrnl.exe)

SystemCodeIntegrityInformation:
        - IntegrityOptions: 0x0203
           0x0001: CODEINTEGRITY_OPTION_ENABLED
           0x0002: CODEINTEGRITY_OPTION_TESTSIGN
           0x0200: CODEINTEGRITY_OPTION_FLIGHTING_ENABLED

SystemKernelDebuggerInformation:
        - KernelDebuggerEnabled: 1
        - KernelDebuggerNotPresent: 1

SystemKernelDebuggerInformationEx:
        - DebuggerAllowed: 1
        - DebuggerEnabled: 1
        - DebuggerPresent: 0

SharedUserData->KdDebuggerEnabled: 0x01

SystemKernelDebuggerFlags: 0x00

SystemCodeIntegrityPolicyInformation:
        - Options: 0x40000000
        - HVCIOptions: 0x0000

SystemIsolatedUserModeInformation:
        - SecureKernelRunning: 0
        - HvciEnabled: 0
        - HvciStrictMode: 0
        - DebugEnabled: 0
        - FirmwarePageProtection: 0
        - EncryptionKeyAvailable: 0
        - TrustletRunning: 0
        - HvciDisableAllowed: 0

3d8kAbdul, 2023-11-13:
@Jack3De Могу подсказать другой обход, не использующий EFI , но там бывают BSOD))

cheboxarov, 2024-02-20:
Querying g_CiOptions value...
CI!g_CiOptions at 0xFFFFF803775E3004.
Failure: NtSetSystemEnvironmentValueEx error 0xC000000D
The EfiGuard DXE driver is either not loaded in SETVARIABLE_HOOK mode, or it is malfunctioning.
AdjustCiOptions failed: 0xC000000D

Mattiwatti, 2024-02-20:
@cheboxarov This is a different error status than the one in the OP. The error message already tells you what the issue is.

@3d8kAbdul Please limit your posts to English only, so that others who don't speak Russian (including myself...) can also read them.