Mattiwatti/EfiGuard #90: help me , get some errors here

vikkkkkrie, 2023-05-24:
i am already change to uefi 
hope you can help me please sir , Thanks before 
iam using tutorial in this video : https://www.youtube.com/watch?v=_auePp1nTHs
to delete loader.config.efi  ( maybe this is the problem )
 

[image]
[image]

Mattiwatti, 2023-05-24:
You're aware that the video you posted wasn't made by me, right?

You're seeing the error you're seeing because you're trying to boot a legacy (BIOS/MBR) OS. **Efi**Guard depends on the EFI boot and runtime services for operation.

If you want to convert an existing Windows installation to a UEFI/GPT style installation, there are a few tools to do this, such as [MBR2GPT](https://answers.microsoft.com/en-us/windows/forum/all/convert-an-existing-windows-10-installation-from/aa8c2de3-460b-4a8c-b30b-641405f800d7).

vikkkkkrie, 2023-05-25:
> Anda sadar bahwa video yang Anda posting bukan buatan saya, bukan?
> 
> Anda melihat kesalahan yang Anda lihat karena Anda mencoba mem-boot OS lama (BIOS/MBR). **Efi** Guard bergantung pada layanan boot dan runtime EFI untuk pengoperasian.
> 
> Jika Anda ingin mengonversi penginstalan Windows yang ada ke penginstalan gaya UEFI/GPT, ada beberapa alat untuk melakukannya, seperti [MBR2GPT](https://answers.microsoft.com/en-us/windows/forum/all/convert-an-existing-windows-10-installation-from/aa8c2de3-460b-4a8c-b30b-641405f800d7) .

thank u sir , it's works 
btw can i ask u for other think ?

why my Cheat Engine Still error when i check EXTRA , cause iam already create the sys file in Visual Studio

[image]

[image]